MIT-SPARK/TEASER-plusplus #1: python/README.md bugs

thduynguyen, 2020-01-24:
1. Small typo in target name:
```
make teaserpp-python -j4
```
Should be 
```
make teaserpp_python -j4
```

2. Segfault when running the example in `python/README.md`
```
TEASER++ Parameters are: <Params with noise_bound=0.01
cbar2=1
estimate_sccaling=0
rotation_estimation_algorithm=GNC-TLS
>
[pmc heuristic]	 mc = 0
Created adjacency matrix in 3.8147e-06 seconds
[pmc: initial k-core pruning]  before pruning: |V| = 20, |E| = 0
[pmc: initial k-core pruning]  after pruning:  |V| = 20, |E| = 0
[pmc]  initial pruning took 1.69277e-05 sec
[pmc: sorting neighbors]  |E| = 0, |E_sorted| = 0
|V| = 20
*** [pmc: thread 1]   current max clique = 1,  time = 0.00155807 sec>>> [pmc: thread 
[pmc: upper bound reached]  omega = 1
[pmc: bounds updated - thread 15]  time = 0.00159502 sec, |V| = 10 (10 / 20), |E| = 0, w = 1, p = 0, d_min = 0, d_avg = 0, d_max = 0, k_max = 0
[pmc: bounds updated - thread 13]  time = 0.00281596 sec, |V| = 9 (11 / 20), |E| = 0, w = 1, p = 0, d_min = 0, d_avg = 0, d_max = 0, k_max = 0
16]
[pmc: bounds updated - thread 16]  time = 0.0031271 sec, |V| = 16 (4 / 20), |E| = 0, w = 1, p = 0, d_min = 0, d_avg = 0, d_max = 0, k_max = 0
[pmc: bounds updated - thread 14]  time = 0.00323701 sec, |V| = 12 (8 / 20), |E| = 0, w = 1, p = 0, d_min = 0, d_avg = 0, d_max = 0, k_max = 0
-----------------------------------------------------------------------
[1]    32144 segmentation fault (core dumped)  python teaserpp_example2.py

```

3. Small bug when running `teaserpp_example.py`: 
```
Scale outliers (TIM pairs) are:
Note: they should include the outlier points.
Traceback (most recent call last):
  File "teaserpp_example.py", line 51, in <module>
    for i in range(scale_inliers.shape[0]):
AttributeError: 'list' object has no attribute 'shape'
```

jingnanshi, 2020-01-24:
Hi,

Thanks for the feedback. Will update the target name & example code.

For the seg fault, try running `OMP_NUM_THREADS=12 python teaserpp_example2.py`. I've noticed some issues with segmentation faults in the maximum clique library we are using when the number of threads exceed 12.

jingnanshi, 2020-01-24:
The example script is fixed & README has been updated. Let me know if it works!

thduynguyen, 2020-01-24:
Thanks for the quick fix! That works. But the example script in README reports different results for different runs and the results are not the ground truth solution. I guess `solver_params.estimate_scaling` must be `True` instead of `False` as in the example?

thduynguyen, 2020-01-24:
Confirm it works if I change that setting to `True` to match with the example file.

closed 2020-01-24 by thduynguyen

jingnanshi, 2020-01-24:
Yes that's correct. Just updated the README. Good catch!