MIT-SPARK/TEASER-plusplus #2: Documentation

lucasamparo, 2020-01-27:
I can't find the docs on the cmake build folder. Someone could help me?
I used the steps from the README.md

Thanks.

jingnanshi, 2020-01-27:
Try these steps:
1. Make a directory called `build` in your cloned repo
2. In the directory, run `cmake ..`
3. After that, you should see a `doc` folder.

lucasamparo, 2020-01-27:
```
lucasamparo@amparo-N8xxEP6:build (master)$ make doc
/bin/sh: 1: cd: can't cd to /doc
doc/CMakeFiles/doc.dir/build.make:57: recipe for target 'doc/CMakeFiles/doc' failed
make[3]: *** [doc/CMakeFiles/doc] Error 2
CMakeFiles/Makefile2:690: recipe for target 'doc/CMakeFiles/doc.dir/all' failed
make[2]: *** [doc/CMakeFiles/doc.dir/all] Error 2
CMakeFiles/Makefile2:697: recipe for target 'doc/CMakeFiles/doc.dir/rule' failed
make[1]: *** [doc/CMakeFiles/doc.dir/rule] Error 2
Makefile:331: recipe for target 'doc' failed
make: *** [doc] Error 2
```

jingnanshi, 2020-01-27:
Did you run `make doc` in the `build` folder, or in the `build/doc` folder?

lucasamparo, 2020-01-27:
Both. Same results.
```
lucasamparo@amparo-N8xxEP6:doc (master)$ cd ..
lucasamparo@amparo-N8xxEP6:build (master)$ make doc
/bin/sh: 1: cd: can't cd to /doc
doc/CMakeFiles/doc.dir/build.make:57: recipe for target 'doc/CMakeFiles/doc' failed
make[3]: *** [doc/CMakeFiles/doc] Error 2
CMakeFiles/Makefile2:690: recipe for target 'doc/CMakeFiles/doc.dir/all' failed
make[2]: *** [doc/CMakeFiles/doc.dir/all] Error 2
CMakeFiles/Makefile2:697: recipe for target 'doc/CMakeFiles/doc.dir/rule' failed
make[1]: *** [doc/CMakeFiles/doc.dir/rule] Error 2
Makefile:331: recipe for target 'doc' failed
make: *** [doc] Error 2
lucasamparo@amparo-N8xxEP6:build (master)$ sudo make doc
[sudo] senha para lucasamparo: 
/bin/sh: 1: cd: can't cd to /doc
doc/CMakeFiles/doc.dir/build.make:57: recipe for target 'doc/CMakeFiles/doc' failed
make[3]: *** [doc/CMakeFiles/doc] Error 2
CMakeFiles/Makefile2:690: recipe for target 'doc/CMakeFiles/doc.dir/all' failed
make[2]: *** [doc/CMakeFiles/doc.dir/all] Error 2
CMakeFiles/Makefile2:697: recipe for target 'doc/CMakeFiles/doc.dir/rule' failed
make[1]: *** [doc/CMakeFiles/doc.dir/rule] Error 2
Makefile:331: recipe for target 'doc' failed
make: *** [doc] Error 2
lucasamparo@amparo-N8xxEP6:build (master)$ cd doc
lucasamparo@amparo-N8xxEP6:doc (master)$ make doc
/bin/sh: 1: cd: can't cd to /doc
doc/CMakeFiles/doc.dir/build.make:57: recipe for target 'doc/CMakeFiles/doc' failed
make[3]: *** [doc/CMakeFiles/doc] Error 2
CMakeFiles/Makefile2:690: recipe for target 'doc/CMakeFiles/doc.dir/all' failed
make[2]: *** [doc/CMakeFiles/doc.dir/all] Error 2
CMakeFiles/Makefile2:697: recipe for target 'doc/CMakeFiles/doc.dir/rule' failed
make[1]: *** [doc/CMakeFiles/doc.dir/rule] Error 2
Makefile:172: recipe for target 'doc/CMakeFiles/doc.dir/rule' failed
make: *** [doc/CMakeFiles/doc.dir/rule] Error 2
lucasamparo@amparo-N8xxEP6:doc (master)$ sudo make doc
/bin/sh: 1: cd: can't cd to /doc
doc/CMakeFiles/doc.dir/build.make:57: recipe for target 'doc/CMakeFiles/doc' failed
make[3]: *** [doc/CMakeFiles/doc] Error 2
CMakeFiles/Makefile2:690: recipe for target 'doc/CMakeFiles/doc.dir/all' failed
make[2]: *** [doc/CMakeFiles/doc.dir/all] Error 2
CMakeFiles/Makefile2:697: recipe for target 'doc/CMakeFiles/doc.dir/rule' failed
make[1]: *** [doc/CMakeFiles/doc.dir/rule] Error 2
Makefile:172: recipe for target 'doc/CMakeFiles/doc.dir/rule' failed
make: *** [doc/CMakeFiles/doc.dir/rule] Error 2
```

jingnanshi, 2020-01-27:
What operating system are you using?

lucasamparo, 2020-01-27:
Ubuntu 16.04 LTS

jingnanshi, 2020-01-27:
Make sure you have `doxygen` installed, and then run `doxygen` in `build/doc` folder.

lucasamparo, 2020-01-27:
```
lucasamparo@amparo-N8xxEP6:doc (master)$ doxygen
warning: ignoring unsupported tag `TOC_INCLUDE_HEADINGS   =' at line 313, file Doxyfile
warning: ignoring unsupported tag `PLANTUML_CFG_FILE      =' at line 2429, file Doxyfile
warning: tag INPUT: input source `/teaser' does not exist
warning: tag INPUT: input source `/doc' does not exist
error: tag OUTPUT_DIRECTORY: Output directory `/doc' does not exist and cannot be created
Exiting...
```

lucasamparo, 2020-01-27:
When I ran ```doxygen doc``` at the build folder, created a html folder and place the documentation here....
But, without any content LOL

[image]

jingnanshi, 2020-01-27:
Sorry, I can't reproduce the error on my side. What folders are there in your `build` folder after you run `cmake ..`?

nkhedekar, 2020-01-27:
I am having the same issue on an Ubuntu 16.04.6 system.

Build folder contents: 

```
.
├── CMakeCache.txt
├── CMakeFiles
│   ├── 3.16.3
│   │   ├── CMakeCCompiler.cmake
│   │   ├── CMakeCXXCompiler.cmake
│   │   ├── CMakeDetermineCompilerABI_C.bin
│   │   ├── CMakeDetermineCompilerABI_CXX.bin
│   │   ├── CMakeSystem.cmake
│   │   ├── CompilerIdC
│   │   │   ├── a.out
│   │   │   ├── CMakeCCompilerId.c
│   │   │   └── tmp
│   │   └── CompilerIdCXX
│   │       ├── a.out
│   │       ├── CMakeCXXCompilerId.cpp
│   │       └── tmp
│   ├── cmake.check_cache
│   ├── CMakeDirectoryInformation.cmake
│   ├── CMakeError.log
│   ├── CMakeOutput.log
│   ├── CMakeRuleHashes.txt
│   ├── CMakeTmp
│   ├── FindOpenMP
│   │   ├── ompver_C.bin
│   │   ├── ompver_CXX.bin
│   │   ├── OpenMPCheckVersion.c
│   │   ├── OpenMPCheckVersion.cpp
│   │   ├── OpenMPTryFlag.c
│   │   └── OpenMPTryFlag.cpp
│   ├── Makefile2
│   ├── Makefile.cmake
│   ├── progress.marks
│   └── TargetDirectories.txt
├── cmake_install.cmake
├── CTestTestfile.cmake
├── doc
│   ├── CMakeFiles
│   │   ├── CMakeDirectoryInformation.cmake
│   │   ├── doc.dir
│   │   │   ├── build.make
│   │   │   ├── cmake_clean.cmake
│   │   │   ├── DependInfo.cmake
│   │   │   └── progress.make
│   │   └── progress.marks
│   ├── cmake_install.cmake
│   ├── CTestTestfile.cmake
│   ├── Doxyfile
│   └── Makefile
├── googletest-build
│   ├── CMakeFiles
│   │   ├── CMakeDirectoryInformation.cmake
│   │   └── progress.marks
│   ├── cmake_install.cmake
│   ├── CTestTestfile.cmake
│   ├── googlemock
│   │   ├── CMakeFiles
│   │   │   ├── CMakeDirectoryInformation.cmake
│   │   │   ├── gmock.dir
│   │   │   │   ├── build.make
│   │   │   │   ├── cmake_clean.cmake
│   │   │   │   ├── cmake_clean_target.cmake
│   │   │   │   ├── DependInfo.cmake
│   │   │   │   ├── depend.make
│   │   │   │   ├── flags.make
[... 1101 lines cut ...]
│           └── pmc-gitupdate.cmake
└── tinyply-src
    ├── assets
    │   ├── bunny.ply
    │   ├── elephant.ply
    │   ├── icosahedron_ascii.ply
    │   ├── icosahedron.ply
    │   ├── sofa_ascii.ply
    │   └── sofa.ply
    ├── BUILD
    ├── cmake
    │   ├── Config.cmake.in
    │   └── tinyply.pc.in
    ├── CMakeLists.txt
    ├── readme.md
    ├── source
    │   ├── example.cpp
    │   ├── Makefile
    │   ├── tinyply.cpp
    │   └── tinyply.h
    └── WORKSPACE

191 directories, 995 files

```

jingnanshi, 2020-01-27:
Okay I know what's the problem now. Working on a fix.

jingnanshi, 2020-01-27:
Can you guys test whether this branch fixes it: https://github.com/MIT-SPARK/TEASER-plusplus/tree/hotfix/fix-doc ? Thanks

nkhedekar, 2020-01-28:
@jingnanshi This works for me. Thank you! :slightly_smiling_face:

lucasamparo, 2020-01-28:
It's fine now. Thanks buddy.

lucasamparo, 2020-01-28:
Sorry about the delay, I'm from Brazil hehe

lucasamparo, 2020-01-28:
I think a Pull-Request could be opened now.

lucasamparo, 2020-01-28:
A last question: Have you a CMake example to compile a code using TeaserCPP?

jingnanshi, 2020-01-28:
I'll put one in the README. I'll let you know when.

nkhedekar, 2020-01-28:
This might be a personal opinion but can we keep this issue open until the pull request is merged so that others don't have to look for a closed issue?

lucasamparo, 2020-01-28:
> This might be a personal opinion but can we keep this issue open until the pull request is merged so that others don't have to look for a closed issue?

You're right. Reopening the issue.

PR #5 (2020-01-28): Fix documentation building error
Fix the CMake variables to be consistent with project names.

Fixes #2
commit a2dfff7d (2020-02-03): Syntax fix #2
closed 2020-01-28 by jingnanshi