MIT-SPARK/TEASER-plusplus #3: Segmentation fault on python2.7

ChristopherLu, 2020-01-28:
Using the solver parameters given by the tutorial example on my own data, meeting the following error:

`
[pmc heuristic]	 mc = 0
Created adjacency matrix in 1.88351e-05 seconds
[pmc: initial k-core pruning]  before pruning: |V| = 142, |E| = 0
[pmc: initial k-core pruning]  after pruning:  |V| = 142, |E| = 0
[pmc]  initial pruning took 1.28746e-05 sec
[pmc: sorting neighbors]  |E| = 0, |E_sorted| = 0
|V| = 142
*** [pmc: thread 37]   current max clique = 1,  time = >>> [pmc: thread [pmc: bounds updated - thread [pmc: bounds updated - thread [pmc: bounds updated - thread 0.0017879 sec
46[pmc: upper bound reached]  omega = 1
[pmc: bounds updated - thread 18]  time = 0.00187087 sec, |V| = 139 (3 / 142), |E| = 0, w = 1, p = 0, d_min = 0, d_avg = 0, d_max = 0, k_max = 0`
`Segmentation fault (core dumped)`

Any hints on what happened :(

Thx!

jingnanshi, 2020-01-28:
Try run `OMP_NUM_THREADS=12 python test.py`. The PMC library seems to have seg faults when the number of threads exceed 12

ChristopherLu, 2020-01-28:
Hi, 

Did what you suggested, but got new error on pmc upper bound:

`[pmc heuristic]	 mc = 0
Created adjacency matrix in 2.09808e-05 seconds
[pmc: initial k-core pruning]  before pruning: |V| = 117, |E| = 0
[pmc: initial k-core pruning]  after pruning:  |V| = 117, |E| = 0
[pmc]  initial pruning took 1.00136e-05 sec
[pmc: sorting neighbors]  |E| = 0, |E_sorted| = 0
|V| = 117
*** [pmc: thread 12]   current max clique = 1,  time = 7.82013e-05 sec
[pmc: upper bound reached]  omega = 1
Segmentation fault (core dumped)`

Should I change some hyper-parameter?

ChristopherLu, 2020-01-28:
My current setting:

_solver_params = teaserpp_python.RobustRegistrationSolver.Params()
solver_params.cbar2 = 1
solver_params.noise_bound = 1
solver_params.estimate_scaling = True
solver_params.rotation_estimation_algorithm = (
    teaserpp_python.RobustRegistrationSolver.ROTATION_ESTIMATION_ALGORITHM.GNC_TLS
)
solver_params.rotation_gnc_factor = 1.4
solver_params.rotation_max_iterations = 100
solver_params.rotation_cost_threshold = 1e-12_

jingnanshi, 2020-01-28:
Interesting, can you show me your input data? It looks like it can't find a max clique larger than 1.

ChristopherLu, 2020-01-28:
Sure, FYI, my data is very sparse and noisy. 

How do you want to me to send the data sample?

jingnanshi, 2020-01-28:
jnshi@mit.edu

ChristopherLu, 2020-01-31:
It just occurred to me that TEASER is a correspondence based method, right? The points clouds of my data are unorganized (i.e., i-th column in frame_t^-1 is not necessarily corresponding to the i-th column in the frame_t). Is that the reason I keep getting error msg? 

Thanks.

jingnanshi, 2020-01-31:
Yes that's correct. TEASER++ takes in point correspondences.

ChristopherLu, 2020-01-31:
Do you have any recommendation to find such correspondences? In the open figure of the TEASER paper, I see there is a version of it can deal with correspondence-free scenarios.

jingnanshi, 2020-01-31:
Yeah, so we deal with correspondence-free scenarios by generating all possible matches between the two point clouds. The Python binding won't do this automatically for you, but you can use the following snippet:
```
INDICES = np.transpose(
    [
        np.tile(
            range(NUM_SAMPLES_FOR_SRC),
            len(range(NUM_SAMPLES_FOR_DST)),
        ),
        np.repeat(
            range(NUM_SAMPLES_FOR_DST),
            len(range(NUM_SAMPLES_FOR_SRC)),
        ),
    ]
)
SRC_SAMPLES_INDICES = INDICES[:, 0]
DST_SAMPLES_INDICES = INDICES[:, 1]
```

Obviously it works only if the point clouds are reasonably sized. On my machine, it terminates in a reasonably amount of time for point clouds of ~100 points (so ~10^4 correspondences). 

Other than that, you can use feature descriptors. We recommend using learned feature descriptors instead of handcrafted ones like FPFH. The one we tested in paper is called 3DSmoothNet, which performs very well in our tests.

ChristopherLu, 2020-02-01:
Works for me. Thanks!

jingnanshi, 2020-02-02:
No problem! Let me know if you need any more clarifications. Closing the issue now.

commit d124cc3c (2020-02-03): Syntax fix #3
closed 2020-02-02 by jingnanshi