MIT-SPARK/TEASER-plusplus #4: CPP Example, using CMake Compilation

lucasamparo, 2020-01-28:
I've started with the code (using the snippet from README.md) but in the compilation, find an error.

<details>
<summary>CMakeLists.txt</summary>
<pre>
cmake_minimum_required(VERSION 3.10)
set (CMAKE_CXX_STANDARD 11)

project(teaserpp_cpp_example)

find_package(teaserpp REQUIRED)
find_package(Eigen3 REQUIRED)

include_directories(
  ${EIGEN3_INCLUDE_DIR}
  ${teaserpp_INCLUDE_DIR}
)

add_executable(cpp_example cpp_example.cpp)
target_link_libraries(cpp_example ${EIGEN3_LIBS} ${teaserpp_LIBS})
</pre>
</details>

<details>
<summary>cpp_example.cpp</summary>
<pre>
#include &lt;iostream&gt;
#include &lt;cstring&gt;
#include &lt;Eigen/Core&gt;
#include &lt;teaser/registration.h&gt;
<br>
int main(int argc, char ** argv) {
if(argc < 3) {
    std::cout << "Incorrect Use." << std::endl;
    std::cout << "./cpp_example <number_of_points> <number_of_outliers>" << std::endl;
    return -1;
}
<br>  
// Total of points
int N = std::stoi(argv[1]);
<br>
// Total of outliers
int O = std::stoi(argv[2]);
<br>
Eigen::Matrix<double, 3, Eigen::Dynamic> src(3, N);
Eigen::Matrix<double, 3, Eigen::Dynamic> dst(3, N + O);
<br>
// Configure your groundtruth transformation
double scale = 1.5;
<br>
Eigen::Matrix3d rotation;
rotation << 0.98370992, 0.17903344, -0.01618098,
                  -0.04165862, 0.13947877, -0.98934839,
                  -0.17486954, 0.9739059, 0.14466493;
              
Eigen::Vector3d translation(-1, 0, 1);
<br>  
// Populate src & dst with your correspondences ...
for(int i = 0; i < N; i++) {
Eigen::Vector3d v = Eigen::Vector3d::Random();
<br>
src.col(i) = v;
dst.col(i) = scale * (rotation * v) + translation;
}
<br>
// Create your outliers
for(int i = 0; i < O; i++) {
    dst.col(N + i) = Eigen::Vector3d::Random();
}
<br>
// Populate solver parameters
teaser::RobustRegistrationSolver::Params params;
params.cbar2 = 1;
params.noise_bound = 0.01;
params.estimate_scaling = false;
params.rotation_estimation_algorithm =
      teaser::RobustRegistrationSolver::ROTATION_ESTIMATION_ALGORITHM::GNC_TLS;
params.rotation_gnc_factor = 1.4;
params.rotation_max_iterations = 100;
params.rotation_cost_threshold = 1e-6;
<br>
// Initialize solver
teaser::RobustRegistrationSolver solver(params);
<br>
// Solve
solver.solve(src, dst);
<br>
// Get solution
auto solution = solver.getSolution();
<br>
return 0;
</pre>
}
</details>

Error:
```
Scanning dependencies of target cpp_example
[ 50%] Building CXX object CMakeFiles/cpp_example.dir/cpp_example.cpp.o
In file included from /tmp/TEASER-plusplus/cpp/cpp_example.cpp:4:0:
/usr/local/include/teaser/registration.h: In member function ‘void teaser::RobustRegistrationSolver::reset(const teaser::RobustRegistrationSolver::Params&)’:
/usr/local/include/teaser/registration.h:666:11: error: ‘make_unique’ is not a member of ‘std’
           std::make_unique<teaser::TLSScaleSolver>(params_.noise_bound, params_.cbar2));
           ^
/usr/local/include/teaser/registration.h:666:50: error: expected primary-expression before ‘>’ token
           std::make_unique<teaser::TLSScaleSolver>(params_.noise_bound, params_.cbar2));
                                                  ^
/usr/local/include/teaser/registration.h:669:11: error: ‘make_unique’ is not a member of ‘std’
           std::make_unique<teaser::ScaleInliersSelector>(params_.noise_bound, params_.cbar2));
           ^
/usr/local/include/teaser/registration.h:669:56: error: expected primary-expression before ‘>’ token
           std::make_unique<teaser::ScaleInliersSelector>(params_.noise_bound, params_.cbar2));
                                                        ^
/usr/local/include/teaser/registration.h:678:28: error: ‘make_unique’ is not a member of ‘std’
       setRotationEstimator(std::make_unique<teaser::GNCTLSRotationSolver>(rotation_params));
                            ^
/usr/local/include/teaser/registration.h:678:73: error: expected primary-expression before ‘>’ token
       setRotationEstimator(std::make_unique<teaser::GNCTLSRotationSolver>(rotation_params));
                                                                         ^
/usr/local/include/teaser/registration.h:682:28: error: ‘make_unique’ is not a member of ‘std’
       setRotationEstimator(std::make_unique<teaser::FastGlobalRegistrationSolver>(rotation_params));
                            ^
/usr/local/include/teaser/registration.h:682:81: error: expected primary-expression before ‘>’ token
       setRotationEstimator(std::make_unique<teaser::FastGlobalRegistrationSolver>(rotation_params));
                                                                                 ^
/usr/local/include/teaser/registration.h:689:9: error: ‘make_unique’ is not a member of ‘std’
         std::make_unique<teaser::TLSTranslationSolver>(params_.noise_bound, params_.cbar2));
         ^
/usr/local/include/teaser/registration.h:689:54: error: expected primary-expression before ‘>’ token
         std::make_unique<teaser::TLSTranslationSolver>(params_.noise_bound, params_.cbar2));
                                                      ^
CMakeFiles/cpp_example.dir/build.make:62: recipe for target 'CMakeFiles/cpp_example.dir/cpp_example.cpp.o' failed
make[3]: *** [CMakeFiles/cpp_example.dir/cpp_example.cpp.o] Error 1
CMakeFiles/Makefile2:75: recipe for target 'CMakeFiles/cpp_example.dir/all' failed
make[2]: *** [CMakeFiles/cpp_example.dir/all] Error 2
CMakeFiles/Makefile2:87: recipe for target 'CMakeFiles/cpp_example.dir/rule' failed
make[1]: *** [CMakeFiles/cpp_example.dir/rule] Error 2
Makefile:118: recipe for target 'cpp_example' failed
make: *** [cpp_example] Error 2
```

jingnanshi, 2020-01-28:
You need to set the standard to at least C++14

lucasamparo, 2020-01-28:
Undefined reference now.
I've forgotten some linking?

jingnanshi, 2020-01-28:
Try something like this: `target_link_libraries(test teaserpp::teaser_registration)` 
Also, since you are scaling, make sure to use `estimate_scaling=true` in the parameters.

lucasamparo, 2020-01-28:
Worked.

The line function solver.getSolution() retrieve a instance of teaserpp::teaser::RegistrationSolution?

jingnanshi, 2020-01-28:
Should be just `teaser::RegistrationSolution`. I think I implemented the print function so you should be able to just print it.
 `teaserpp::` is just a CMake namespace, not in the actual code.

lucasamparo, 2020-01-28:
> Should be just `teaser::RegistrationSolution`. I think I implemented the print function so you should be able to just print it.
> `teaserpp::` is just a CMake namespace, not in the actual code.

I know. Just using the "way" from the documentation. Thanks again.
The code is complete now.

Have you configured some CMake variables from the teaserpp?

jingnanshi, 2020-01-28:
No. I use CMake's target-based flow, so just `target_link_libraries(cpp_example teaserpp::teaser_registration)` should be enough.

What variables do you think you need?

lucasamparo, 2020-01-28:
> What variables do you think you need?

Just to keep similar than other libraries, like OpenCV, Eigen or PCL...

Your library looks very very very good, man.

lucasamparo, 2020-01-28:
[cpp_example.cpp](https://pastebin.com/FCEbMvEX)
[CMakeLists.txt](https://pastebin.com/3DJUELQ7)

jingnanshi, 2020-01-28:
Thanks for the compliment & thanks for posting the examples up! 

For now, I'll stick with the target based flow since this library doesn't have a lot of components. But if we add more stuff do it in the future, I'll consider implement something like OpenCV. Thanks for the feedback. I'll close this issue now.

closed 2020-01-28 by jingnanshi

lucasamparo, 2020-01-28:
Feel free to use the example in your repository :smile: 
The prior results here sounds very good.

"When I'm a grown-up man I want to be just like you."