MIT-SPARK/TEASER-plusplus #10: Random Segmentation Fault when solver_params.estimate_scaling = False

georgegu1997, 2020-02-06:
HI, thanks for the amazing paper and software. 

I am using the python binding of `teaserpp_python.RobustRegistrationSolver`, but get `Segmentation fault (core dumped)` sometimes when I set `solver_params.estimate_scaling = False` and the shape of `src` and `dst` mismatches. The following code can produce the error and it may or may not throw the error at different runs. 

```
import numpy as np
import teaserpp_python

# Generate random data points
src = np.random.rand(3, 20)

# Apply arbitrary scale, translation and rotation
scale = 1.0
translation = np.array([[1], [0], [-1]])
rotation = np.array([[0.98370992, 0.17903344, -0.01618098],
                     [-0.04165862, 0.13947877, -0.98934839],
                     [-0.17486954, 0.9739059, 0.14466493]])
dst = scale * np.matmul(rotation, src) + translation
dst = np.concatenate([dst, 2.0*np.random.rand(3, 20)], axis=1)

# Add two outliers
dst[:, 1] += 10
dst[:, 9] += 15

print(src.shape, src.dtype)
print(dst.shape, dst.dtype)

# Populate the parameters
solver_params = teaserpp_python.RobustRegistrationSolver.Params()
solver_params.cbar2 = 1
solver_params.noise_bound = 0.01
solver_params.estimate_scaling = False
solver_params.rotation_estimation_algorithm = (
    teaserpp_python.RobustRegistrationSolver.ROTATION_ESTIMATION_ALGORITHM.GNC_TLS
)
solver_params.rotation_gnc_factor = 1.4
solver_params.rotation_max_iterations = 100
solver_params.rotation_cost_threshold = 1e-12
print("TEASER++ Parameters are:", solver_params)
teaserpp_solver = teaserpp_python.RobustRegistrationSolver(solver_params)

solver = teaserpp_python.RobustRegistrationSolver(solver_params)
solver.solve(src, dst)

solution = solver.getSolution()

# Print the solution
print("Solution is:", solution)
```

This issue disappears if `solver_params.estimate_scaling = True` but in rare cases it throws `MemoryError: std::bad_alloc`. I am already using `OMP_NUM_THREADS=12`. 

Could you please look into this case and see why this happens? I am using Ubuntu 18.04 with Python 3.6.10.

georgegu1997, 2020-02-06:
Ah, sorry, I misunderstood the input.

closed 2020-02-06 by georgegu1997

TobiasFischerP4D, 2020-03-23:
I seem to have a similar misconception. Would you mind explaining? :)

jingnanshi, 2020-03-24:
Hi @TobiasFischerP4D, thanks for your interest. `src` and `dst` need to have matching sizes (coming from a feature matching algorithm). Let me know if you have any other questions.

TobiasFischerP4D, 2020-03-25:
So, the data should be structured such that the first point in `src` corresponds to the first point in `dst` and so on. Why does this only need to be the case if we want to solve for scale?

jingnanshi, 2020-03-25:
`src` and `dst` need to be correspondences no matter whether you want to estimate scale or not. The fact that it doesn't crash when you set `solver_params.estimate_scaling = True` is just undefined behavior (probably because Eigen turns off size checking for release build).