MIT-SPARK/TEASER-plusplus #11: Data maximum size

ehocquard, 2020-02-13:
Hi,

First of all, many thanks for the work that was done on this, and the quality of code/documentation/example is impressive.

We are evaluating it to see if it fits our needs.
We are trying it to register two Kinect dataset.

Sadly, we did not have a single successful registration so far.

One problem we have is the size of the data that can be handled, it fails (i.e. bad alloc) if N above (around) 30K points.
Did you encounter this issue? 
What is the maximum size you tried?
On theory, it seems like it could handle big N, in practice no, I could explore why if needed, I think it is related to one of eigen operations.

Also the amount of memory usage, on registering two dataset with N=30K, it uses around 50Gb of ram memory, is this normal? 

Again many thanks for the work and sharing!
Etienne

jingnanshi, 2020-02-13:
Hi Etienne,

Thanks for letting us know. 

TEASER++ is not a correspondence-free method. That is to say, you need to first use some sort of feature extraction algorithms to extract features from the raw point clouds. You then match the features, and use the matched points as inputs to TEASER++. 

I assume N=30K is the original size of the point cloud. You should be able to use some learned descriptors to match some sampled points in feature space, and then use TEASER++ to solve the registration.

With N=30K, 50Gb of RAM is normal. In our testing, we usually just sample ~100 to a few thousand feature-matched points. The maximum number of correspondence inputs I've tried is ~40k to 50k correspondences, which takes ~60 GB to 70 GB of RAM.

jingnanshi, 2020-02-13:
In terms of good 3D descriptors, I find the performance of handcrafted algorithms such as FPFH unreliable on real-world datasets. I recommend using learned 3D descriptors for your project.

ehocquard, 2020-02-14:
Hi,

I see, it makes sense and I should have think about it tbh.

We'll look into it and let you know of the result.

Feel free to close the issue.

Many thanks!

jingnanshi, 2020-02-14:
Feel free to email me (jnshi@mit.edu) if you want me to take a look at the data.

monage2018, 2020-03-17:
@ehocquard Hi,I have the same problem with you,and now, i notice that The TEASER needs Correspondence input when the original size of the point cloud is so big.So, How are you doing? what feature-extraction-algorithms do you use?Forgive my poor English！

jingnanshi, 2020-03-17:
@monage2018 Hi, TEASER++ is a correspondence-based method. So ideally you should generate correspondences by feature matching. You can use 3DSmoothNet (which we used in our paper) to generate correspondences. If the point clouds are small, you can generate all possible correspondences, then feed those correspondences to TEASER++.

closed 2020-03-25 by jingnanshi