MIT-SPARK/TEASER-plusplus #12: Issues Installing the TEASER++ Python binding

HabibaZImam, 2020-02-23:
Hello jingnan,

I am working on point cloud registration for reverse engineering and additive repair, for which your TEASER++ code looks very promising and capable. I am trying to run it in python. I do not have much experience with C++ or Cmake, therefore I'm hoping that you can help me with the final step to install TEASER++ to my Anaconda environment.
I downloaded the TEASER++ repository and ran it through Cmake. For that I had to manually download pmc, pybind11 and tinyply and place them into the '...-src' folders inside the build folder, because Cmake was not able to install them by itself. After doing that I now have the 'build/python' folder which you can see here.
[image]

When I run 'pip install .' in this folder I get the following error. 
```
Processing c:\users\habib\desktop\teaser-plusplus-master\build\python
    ERROR: Command errored out with exit status 1:
     command: 'c:\users\habib\anaconda3\envs\ipk\python.exe' -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\habib\\AppData\\Local\\Temp\\pip-req-build-y6hagk_w\\setup.py'"'"'; __file__='"'"'C:\\Users\\habib\\AppData\\Local\\Temp\\pip-req-build-y6hagk_w\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base 'C:\Users\habib\AppData\Local\Temp\pip-req-build-y6hagk_w\pip-egg-info'
         cwd: C:\Users\habib\AppData\Local\Temp\pip-req-build-y6hagk_w\
    Complete output (7 lines):
    running egg_info
    creating C:\Users\habib\AppData\Local\Temp\pip-req-build-y6hagk_w\pip-egg-info\teaserpp_python.egg-info
    writing C:\Users\habib\AppData\Local\Temp\pip-req-build-y6hagk_w\pip-egg-info\teaserpp_python.egg-info\PKG-INFO
    writing dependency_links to C:\Users\habib\AppData\Local\Temp\pip-req-build-y6hagk_w\pip-egg-info\teaserpp_python.egg-info\dependency_links.txt
    writing top-level names to C:\Users\habib\AppData\Local\Temp\pip-req-build-y6hagk_w\pip-egg-info\teaserpp_python.egg-info\top_level.txt
    writing manifest file 'C:\Users\habib\AppData\Local\Temp\pip-req-build-y6hagk_w\pip-egg-info\teaserpp_python.egg-info\SOURCES.txt'
    error: package directory 'C:Users\habib\Desktop\TEASER-plusplus-master\build\python\teaserpp_python' does not exist
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
```

Does someone know what I'm doing wrong?
I'd be thankful for any hints.

jingnanshi, 2020-02-24:
Hi,

Just from the message itself I can't tell what's wrong. I haven't tested on Windows before so I am not familiar with what issues might occur. I'll try to build TEASER++ on Windows this week and get back to you.

HabibaZImam, 2020-02-24:
I'd really appreciate that. Meanwhile, I'm going to try it in ubuntu.

jingnanshi, 2020-02-28:
Hi, I tried with windows, and unfortunately I'm not familiar with C++/Python development on windows to figure this out in a reasonably amount of time. Please stick with Ubuntu as much as you can. Alternatively, you can also use docker on windows.

jingnanshi, 2020-03-02:
Closing this for now. Please reopen if you have more questions.

closed 2020-03-02 by jingnanshi