MIT-SPARK/TEASER-plusplus #15: Example for using with point clouds

esteimle, 2020-03-17:
Could you share an example of how to run this against a pair of point clouds?  Like from a ply or obj file or something? The examples just show making a random array and using that as the source.

Thanks!

jingnanshi, 2020-03-17:
Hi @esteimle, thanks for your interest. Are you interested in a C++ example or a Python example?

esteimle, 2020-03-18:
Hi, thanks for writing back.  I'm interested in C++ example if possible.  I have some 3d point clouds I captured with my device from different angles and I want to learn how to combine them into one model.  I read your documentation saying I should first use 3Dsmooth net to get correspondances, so I'm going to try to learn about how to do that today. Then I hope to use teaser to combine the point clouds into one complete model.   I also tried to make my own sample program based off of your tests in c++ but I don't have the linking right yet for the build :)

jingnanshi, 2020-04-01:
@esteimle please take a look at PR #20 . New examples are added in the `examples/` folder. Let me know if you have any suggestions for improvements!

esteimle, 2020-04-02:
Great thank you!

jingnanshi, 2020-04-03:
Merged #20 . Closing issue now.

closed 2020-04-03 by jingnanshi