MIT-SPARK/TEASER-plusplus #17: Cannot reproduce numbers on the 3DMatch test set.

chrischoy, 2020-03-25:
Hi Hank,

I am trying to evaluate TEASER on 3DMatch, but I could not get numbers even close to 99% on 3DMatch test set. In fact, the numbers that I got is around 40%. For some challenging pairs, TEASER took almost 1000sec.

Could you post your 3DMatch pipeline?

jingnanshi, 2020-03-25:
Hi Chris,

Thank you for your interest. Are you using 3DSmoothNet? Also, what are the parameters you are using?

I'm planning to post an example showing how to incorporate 3DSmoothNet with TEASER++ using Python, so stay tuned.

chrischoy, 2020-03-25:
Hi Jingnan,

I am using it with FCGF. I'll post my registration setup on the repo soon as well.

closed 2020-03-26 by chrischoy

XuyangBai, 2020-05-13:
Hi @jingnanshi 

Thanks for your sharing. I am also trying TEASER++ on 3DMatch, would you mind sharing the parameters with me?  

Best,
Xuyang.

jingnanshi, 2020-05-13:
@XuyangBai see this file: https://github.com/MIT-SPARK/TEASER-plusplus/blob/master/examples/teaser_python_3dsmooth/teaser_python_3dsmooth.py The parameters should all be there. Let me know if you have any more questions!

XuyangBai, 2020-05-14:
@jingnanshi  Thanks a lot !