MIT-SPARK/TEASER-plusplus #18: malloc(): largebin double linked list corrupted (nextsize)

chrischoy, 2020-03-25:
I am getting this error `malloc(): largebin double linked list corrupted (nextsize)` intermittently and can't run this for more than 5 registration pairs.

jingnanshi, 2020-03-25:
Haven't seen this before. How large is your RAM?

chrischoy, 2020-03-26:
It is 32G. It seems that FCGF gives too many inliers for TEASER and allocating too much memory.

jingnanshi, 2020-03-27:
For reference: error msg is from here: https://github.com/kraj/glibc/blob/master/malloc/malloc.c#L3868

manhvuvan281, 2021-05-25:
I tried on Intel CPU : OK,
but on AMD CPU: corrupted (malloc).
Anyone have an idea?

manhvuvan281, 2021-05-25:
For AMD CPU users,
Code:
	import os
	os.environ["OMP_NUM_THREADS"] = "4" 
Terminal:
	export OMP_NUM_THREADS=4  #~ Max thread
	#echo $OMP_NUM_THREADS     #show 4
        python ./filename.py

closed 2023-05-22 by jingnanshi