MIT-SPARK/TEASER-plusplus #19: terminate called after throwing an instance of 'std::bad_alloc'

Ao-lei, 2020-03-30:
Hi: 
when use the [PCL](http://pointclouds.org/documentation/tutorials/normal_distributions_transform.php) example data,  I get the error ` terminate called after throwing an instance of 'std::bad_alloc, what():  std::bad_alloc`.
But everything is OK for registration_test/1000point_model.ply。
my code is
```
  std::string file_one = "../data/pcd/room_scan1.pcd";
  std::string file_two = "../data/pcd/room_scan2.pcd";

  pcl::PointCloud<pcl::PointXYZ>::Ptr target_cloud(new pcl::PointCloud<pcl::PointXYZ>);
  if (pcl::io::loadPCDFile<pcl::PointXYZ>(file_one, *target_cloud) == -1) {
    PCL_ERROR ("Couldn't read file 1.pcd \n");
    return (-1);
  }
  Eigen::Matrix<double, 3, Eigen::Dynamic> src(3, target_cloud->size());
  for (size_t i = 0; i < target_cloud->size(); ++i) {
    src.col(i) << target_cloud->points[i].x, target_cloud->points[i].y, target_cloud->points[i].z;
  }
  // Loading second scan of room from new perspective.
  pcl::PointCloud<pcl::PointXYZ>::Ptr input_cloud(new pcl::PointCloud<pcl::PointXYZ>);
  if (pcl::io::loadPCDFile<pcl::PointXYZ>(file_two, *input_cloud) == -1) {
    PCL_ERROR ("Couldn't read 2.pcd \n");
    return (-1);
  }
  Eigen::Matrix<double, 3, Eigen::Dynamic> dst(3, input_cloud->size());
  for (size_t i = 0; i < input_cloud->size(); ++i) {
    dst.col(i) << input_cloud->points[i].x, input_cloud->points[i].y, input_cloud->points[i].z;
  }

  // Start the timer
  auto start = std::chrono::high_resolution_clock::now();
  teaser::RobustRegistrationSolver::Params params;
  params.cbar2 = 1;
  params.noise_bound = 0.01;
  params.estimate_scaling = false;
  params.rotation_estimation_algorithm = teaser::RobustRegistrationSolver::ROTATION_ESTIMATION_ALGORITHM::GNC_TLS;
  params.rotation_gnc_factor = 1.4;
  params.rotation_max_iterations = 100;
  params.rotation_cost_threshold = 1e-6;
  // Prepare the solver object
  teaser::RobustRegistrationSolver solver(params);
  // Solve
  solver.solve(src, dst);
  // Get solution
  auto solution = solver.getSolution();

  // Stop the timer
  auto stop = std::chrono::high_resolution_clock::now();
  auto duration = std::chrono::duration_cast<std::chrono::microseconds>(stop - start);
  std::cout << "Time spent: " << duration.count() << std::endl;
```

prajval10, 2020-04-08:
Hi @Ao-lei and @jingnanshi, I have the same error with my clouds. The error occurs while using the `teaser::RobustRegistrationSolver` `solve()` function.

The error:
```
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
Aborted (core dumped)
```

GDB output:
```
(gdb) 
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
#1  0x00007ffff0a5e801 in __GI_abort () at abort.c:79
#2  0x00007ffff1451957 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#3  0x00007ffff1457ae6 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#4  0x00007ffff1457b21 in std::terminate() () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#5  0x00007ffff1457d54 in __cxa_throw () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#6  0x0000555555573c12 in Eigen::internal::throw_std_bad_alloc() ()
#7  0x00007ffff7bc27f7 in teaser::RobustRegistrationSolver::computeTIMs(Eigen::Matrix<double, 3, -1, 0, 3, -1> const&, Eigen::Matrix<int, 2, -1, 0, 2, -1>*) ()
   from /usr/local/lib/libteaser_registration.so
#8  0x00007ffff7bc7e97 in teaser::RobustRegistrationSolver::solve(Eigen::Matrix<double, 3, -1, 0, 3, -1> const&, Eigen::Matrix<double, 3, -1, 0, 3, -1> const&) ()
   from /usr/local/lib/libteaser_registration.so
#9  0x00007ffff7bc9310 in teaser::RobustRegistrationSolver::solve(teaser::PointCloud const&, teaser::PointCloud const&, std::vector<std::pair<int, int>, std::allocator<std::pair<int, int> > >) () from /usr/local/lib/libteaser_registration.so
#10 0x000055555557112c in main ()
(gdb) quit
```

The example code `teaser_cpp_ply` runs correctly without issues however. Any ideas on how to solve this?

jingnanshi, 2020-04-08:
Hi @prajval10, 

Make sure you are giving TEASER++ **correspondences** instead of raw point clouds. Make sure the two input Eigen matrices have the same dimensions. 

Also, what's your point clouds' sizes?

prajval10, 2020-04-08:
Hi @jingnanshi, thanks for the quick response!

Looking at the source, I believe the cloud sizes need to be the same:
https://github.com/MIT-SPARK/TEASER-plusplus/blob/3f47dd048e5585992fea41089e7dbe89f88509bd/teaser/src/registration.cc#L404

In my case, I am using the TEASER++ correspondences however the sizes are not the same:
```
Object size: 16223
Scene size: 734468
correspondences size: 16303
```
I'm trying to register an object pointcloud extracted from a CAD mesh and a scene point-cloud extracted using a depth sensor which contains the object along with other possible occlusions (similar to http://www.pointclouds.org/documentation/tutorials/correspondence_grouping.php). In this scenario (as in most cases involving registration) the number of points in the object (or source) and the scene (or target) are not the same. So I suppose `teaser::RobustRegistrationSolver` is not meant for such situations?

jingnanshi, 2020-04-08:
Hi @prajval10,

You need one more extra step before TEASER++: using a feature extraction algorithm to extract corresponding feature points from the two point clouds.

In your case, you need to do the following:
1. Get the CAD mesh and scene point cloud
2. Run feature extraction algorithms to extract feature vectors for all points
3. Use a nearest-neighbor algorithm to find matching points in the feature space
4. Feed the correspondences to TEASER++ to solve for registration

For feature extraction, I recommend learned feature descriptors such as [3DSmoothNet](https://github.com/zgojcic/3DSmoothNet) and [FCGF](https://github.com/chrischoy/FCGF).

Let me know if you have more questions.

prajval10, 2020-04-08:
Thanks @jingnanshi ! I am using the `FPFHEstimation` for the features. Here is my **psuedo-code**:

```cpp
// necessary pointers for object and scene clouds
 
// extract normals for object and scene
pcl::NormalEstimationOMP<pcl::PointXYZ, pcl::Normal> norm_est;
norm_est.setKSearch(10.0);
norm_est.setInputCloud(object);
norm_est.compute(*object_normal);

// Extract feature descriptors for object and scene
pcl::FPFHEstimation<pcl::PointXYZ, pcl::Normal, pcl::FPFHSignature33> fpfh;
fpfh.setInputCloud (object);
fpfh.setInputNormals (object_normal);
pcl::search::KdTree<pcl::PointXYZ>::Ptr tree_object (new pcl::search::KdTree<pcl::PointXYZ>);
fpfh.setSearchMethod (tree_object);
fpfh.setRadiusSearch (5.0);
// Compute the features
fpfh.compute (*object_descriptors);

// convert object and scene clouds for pcl::PointCloud<pcl::pointXYZ> to teaser::PointCloud
teaser::PointCloud cloud_out;
for(int i=0; i != pcl_cloud_in.points.size();i++){
        teaser::PointXYZ pt;
        pt.x = pcl_cloud_in.points[i].x;
        pt.y = pcl_cloud_in.points[i].y;
        pt.z = pcl_cloud_in.points[i].z;
        cloud_out.push_back(pt);
    }

teaser::Matcher matcher;
auto correspondences = matcher.calculateCorrespondences(obj_cloud, scene_cloud, *object_descriptor, *scene_descriptor, true, false, false);

// Run TEASER++ registration
// Prepare solver parameters
teaser::RobustRegistrationSolver::Params params;
params.noise_bound = NOISE_BOUND;
params.cbar2 = 1;
params.estimate_scaling = false;
params.rotation_max_iterations = 100;
params.rotation_gnc_factor = 1.4;
params.rotation_estimation_algorithm =
        teaser::RobustRegistrationSolver::ROTATION_ESTIMATION_ALGORITHM::GNC_TLS;
params.rotation_cost_threshold = 0.005;

// Solve with TEASER++
teaser::RobustRegistrationSolver solver(params);
solver.solve(obj_cloud, scene_cloud, correspondences);
auto solution = solver.getSolution();

jingnanshi, 2020-04-08:
@prajval10 can you try this branch: https://github.com/MIT-SPARK/TEASER-plusplus/tree/bugfix/solve_with_correspondences

prajval10, 2020-04-08:
> Edit: @prajval10 can you actually try this branch: https://github.com/MIT-SPARK/TEASER-plusplus/tree/bugfix/correspondence-size. Thanks!

It works now! Thanks :+1:

jingnanshi, 2020-04-08:
Nice to hear! FPFH features don't work very well. If possible, I recommend using learned features. In addition, the `matcher` implemented in TEASER is very rudimentary, and it can certaily be improved.

closed 2020-04-09 by jingnanshi

surajiitd, 2022-11-11:
@prajval10 How did you downsampled the pointclouds. just skipping alternate points will work??

jingnanshi, 2022-11-12:
@surajiitd Thank you for your interest. You can either down sample the point clouds using Open3D's various methods, or use a feature extractor to select salient keypoints (hence it's implicitly conducting downsampling).

surajiitd, 2022-11-12:
thans for your replu @jingnanshi .
I was downsampling kitti pointcloud and registering. but it needs **huge** downsampling (like from 120000 points to just 2000 points).. what if I want to work without downsampling... I was getting badalloc error while doing without downsampling.

xiao10ma, 2025-04-13:
I have more than 1.2 million points, why it limits the number of point clouds??