MIT-SPARK/TEASER-plusplus #22: Teaser and 3D smooth together

esteimle, 2020-04-08:
I finally got the new C examples running and I tried to run them with my own point clouds. But they were too big and caused a Malloc error. Now I understand from reading that what I should do is feed the correspondences in, instead. So I spent some time and now I have 3Dsmoothnet running.  It seems to take in ply files and generate .csv files that I assume are the correspondances?  I'm confused on how to go from pointcloud -> 3Dsmoothnet -> teaser.  Do you have any advice for me?

Thanks!

jingnanshi, 2020-04-08:
The example will be a great place to start. However, the example in the repo does not show you how to generate the SDV files and descriptors from point clouds using 3DSmooth. 

You should take a look at this: https://github.com/zgojcic/3DSmoothNet#instructions To summarize:
1. Use 3DSmooth to generate the SDV parametrization (the csv files you mentioned)
2. Use 3DSmooth to generate descriptors from the SDV files (generate .npz numpy binary files)
3. Use the functions in the example python script to match descriptors & use TEASER++ to generate registration.

esteimle, 2020-04-09:
I was able to run my ply files through 3Dsmooth and then I ran the results through your python demo. I modified the 3dsmooth execute_teaser_global_registration method to print out the estimate rotation and translation values, but they're not working like I expected. It's not registering the two point clouds, in fact it makes the rotation way worse :) 

I went back and used your example data, and I was able to get it to work no problem.  Not sure what I'm missing yet.  Thanks for your help so far.

jingnanshi, 2020-04-09:
@esteimle If you want, you can post your code & data on a git repo / gist and share it with me so I can take a look. My guess is it's either 1) the clouds you are using are not similar to the data used in training for 3DSmooth 2) the feature correspondences calculated are not correct.

esteimle, 2020-04-09:
Sure thank you so much.  Here's a repo with my modifications https://github.com/esteimle/teaser 

in the examples/teaser_python_3dsmooth folder I have a file called eric_3dsmooth.py that I was using.

My data is in examples/example_data/smooth
output_2.ply and output_3.ply are what I'm working on.  They're two depth images of my desk just slightly off angle from one another. I put a picture in this directory called apic.jpg that shows what the ply files look like.

I also tried to use the fphf example you shared, but I always get this error: /ld: cannot find -lteaserpp::teaser_features.  I did turn on FPFH in the main CMake file, and build and make install again.  I had the same problem with teaser_io before but I was able to get around it by using a find library statement pointing to libteaser_io.so, but I don't see the same kind of so for FPFH.

Thank you for your help, I'm still learning all of this :)

fyang064, 2021-07-28:
The 3dmatch_sample in the example_data is a good example to learn and modify. However, I met some troubles when I tried to generate my own data using the 3DSmooth.  
[image]
I assume the "Killed" issue is caused by the exhausted memory but I'm wondering how the author generated the descriptor that had the similar size and number of points as my pointclouds (Less than million vertexes and 5000 keypoints).

Looking forward to hearing from you!

jingnanshi, 2021-07-29:
@fyang064 with feature matching you will have much fewer points left (usually around 500~ from my experience).

closed 2023-05-22 by jingnanshi