MIT-SPARK/TEASER-plusplus #24: Not going really well with the partial view Pointcloud

saadehmd, 2020-05-01:
I am not sure if i got the whole concept wrong, but is this method supposed to work on partial-view pointclouds? The results were not very good when i tried to register a single-view cloud from kinect to it's original 360 degree scanned pointcloud. 
Then instead of this kinect cloud, i just tried cropping the original cloud before transformation, to preserve different levels of the full cloud. I tried from 50-90%. After around 95%, the estimated pose starts getting considerably good and near perfect when i crop nothing out of the cloud.
This made me wonder if i am using the methods in a wrong way.

I also tried inverse-registering, swapping the src <-> destination clouds, so i could register  full - partial instead of partial - full cloud, but that always ends up in segmentation fault. I guess, the code needs the no. of points in src cloud to be either equal to or more than the destination cloud, but fails if there's less.

It's probably a really amateur question. But it would be really helpful if you could give an insight as to what i might be doing wrong.

jingnanshi, 2020-05-02:
Hi @saadehmd, thank you for your interest! TEASER++ is in its essense a **correspondence-based** method. The src/dst clouds need to be the same sizes because they should represent correspondenting points; i.e.: the first column of the dst cloud should correspondent to the first column of the src cloud. 

To generate correspondence, consider 3DSmoothnet or FCGF. Let me know if you have any other questions!

saadehmd, 2020-05-02:
Thanks for your reply. Would it be any useful if I add random outliers in dst cloud just to make the sizes equal. Since it's robust to a large amount of them. Thanks for the other suggestions. I was hoping this works with partial clouds as it was working so perfectly with the full-cloud. Anyways I'll try the other two methods. Is the python example teaser_pythom_3Dsmooth.py for this 3Dsmoothnet method?

jingnanshi, 2020-05-02:
Instead of adding random outliers, it makes more sense to create more correspondences between src and dst points. 

Yes, the example py file shows you how to load 3DSmoothnet descriptors, match them and use TEASER++. To actually generate 3DSmoothnet descriptors, you need to refer to their github repo.

See https://github.com/zgojcic/3DSmoothNet and https://github.com/chrischoy/FCGF.

jingnanshi, 2020-05-03:
Closing this for now. Reopen if you have any more questions.

closed 2020-05-03 by jingnanshi