MIT-SPARK/TEASER-plusplus #25: open3d fpfh

PetropoulakisPanagiotis, 2020-05-03:
Is it possible that we can use open3d fpfh to use solver.solve() like teaser_cpp_fpfh example?

jingnanshi, 2020-05-03:
Yes. TEASER++ doesn't care about what features you use, as long as you have correspondences.

If you follow this tutorial: http://www.open3d.org/docs/release/tutorial/Advanced/global_registration.html, you should be almost all set. You just need to match FPFH features using your own feature matching algorithm.

PetropoulakisPanagiotis, 2020-05-04:
Thank you for your answer. I use open3D frequently and i try to combine it with your API.  I try to import matcher in python like teaser_cpp_fpfh but an error occured(may matcher isn't available in python). Do you have any suggestion ?

jingnanshi, 2020-05-05:
Yes I don't have the matcher binding in Python. You can use Open3D's matcher instead.

closed 2020-06-07 by jingnanshi