MIT-SPARK/TEASER-plusplus #26: Incorrect rotation error calculation

superkirill, 2020-05-14:
The function **bench_utils.get_angular_error(R_gt, R_est)** in the teaser_python_3dsmooth example doesn't compute the rotation error correctly. For example, if R_gt and R_est equal to:

[[ 0.47754719 -0.031702   -0.14449853]
 [ 0.144925    0.00224509  0.47846407]
 [-0.02969151 -0.49892495  0.01133455]]

the function returns  **97.18351924664557**, when it should be **0**, because it is an angle between a rotation matrix and the same rotation matrix. I don't understand why it happens, because the implementation seems to be correct.

jingnanshi, 2020-05-14:
@superkirill Thanks for the feedback! I checked your matrix, and it seems like it's not a proper rotation matrix (its determinant is not 1). If I normalize the matrix to have the determinant be 1, then the function returns correct results.
Here is my Python interpreter output showing the matrix's determinant:
```
>>> import numpy as np
>>> R_gt = np.array([[0.47754719, -0.031702, -0.14449853], [0.144925, 0.00224509, 0.47846407], [-0.02969151,-0.49892495,0.01133455]])
>>> R_gt
array([[ 0.47754719, -0.031702  , -0.14449853],
       [ 0.144925  ,  0.00224509,  0.47846407],
       [-0.02969151, -0.49892495,  0.01133455]])
>>> R_est = R_gt
>>> np.linalg.det(R_gt)
0.12495212468418526
```
And here are the outputs if I normalize the matrix:
```
>>> A = R_gt 
>>> R_gt = A / (np.linalg.det(A))**(1/3)
>>> np.linalg.det(R_gt)
0.9999999999999994
>>> R_gt
array([[ 0.95521635, -0.0634121 , -0.28903396],
       [ 0.28988701,  0.00449075,  0.95705034],
       [-0.0593906 , -0.99797733,  0.02267199]])
>>> R_est=R_gt
>>> R_est
array([[ 0.95521635, -0.0634121 , -0.28903396],
       [ 0.28988701,  0.00449075,  0.95705034],
       [-0.0593906 , -0.99797733,  0.02267199]])
>>> R_gt
array([[ 0.95521635, -0.0634121 , -0.28903396],
       [ 0.28988701,  0.00449075,  0.95705034],
       [-0.0593906 , -0.99797733,  0.02267199]])
>>> bench_utils.get_angular_error(R_gt, R_est)
1.7075472925031877e-06
```
Let me know if you have any more questions!

superkirill, 2020-05-15:
@jingnanshi Thanks for your reply! I indeed didn't think of normalizing the matrix, now everything seems to be fine.

closed 2020-05-15 by superkirill