MIT-SPARK/TEASER-plusplus #28: munmap_chunk(): invalid pointer

LKeaning, 2020-05-25:
Hi, I'm running the minimum python example. When compiling the code everything works fine but when I run the teaser_python_ply.py, I got the following error messages:

==================================================
        TEASER++ Python registration example      
==================================================
*** [pmc heuristic: thread 1]   current max clique = 576,  time = 0.00177217 sec
*** [pmc heuristic: thread 16]   current max clique = 577,  time = 0.0418272 sec
*** [pmc heuristic: thread 4]   current max clique = 578,  time = 0.0511582 sec
*** [pmc heuristic: thread 11]   current max clique = 584,  time = 0.052629 sec
[pmc heuristic]	 mc = 584
Created adjacency matrix in 0.00107789 seconds
[pmc: initial k-core pruning]  before pruning: |V| = 1889, |E| = 324759
[pmc: initial k-core pruning]  after pruning:  |V| = 1889, |E| = 324708
[pmc]  initial pruning took 0.00183296 sec
[pmc: sorting neighbors]  |E| = 649416, |E_sorted| = 649416
|V| = 811
>>> [pmc: thread 16]
[pmc: bounds updated - thread 14]  time = 0.015239 sec, |V| = 1879 (10 / 1889), |E| = 324708, w = 584, p = 0.182091, d_min = 0, d_avg = 343.842, d_max = 810, k_max = 742
[pmc: bounds updated - thread 16]  time = 0.0158432 sec, |V| = 1881 (8 / 1889), |E| = 324708, w = 584, p = 0.182091, d_min = 0, d_avg = 343.842, d_max = 810, k_max = 742
[pmc: bounds updated - thread 15]  time = 0.0159872 sec, |V| = 1877 (12 / 1889), |E| = 324708, w = 584, p = 0.182091, d_min = 0, d_avg = 343.842, d_max = 810, k_max = 742
[pmc: bounds updated - thread 13]  time = 0.016629 sec, |V| = 1874 (15 / 1889), |E| = 324708, w = 584, p = 0.182091, d_min = 0, d_avg = 343.842, d_max = 810, k_max = 742
*** [pmc: thread 8]   current max clique = 585,  time = 0.108998 sec
*** [pmc: thread 10]   current max clique = 586,  time = 0.110153 sec
*** [pmc: thread 4]   current max clique = 589,  time = 0.111155 sec
*** [pmc: thread 6]   current max clique = 591,  time = 0.112657 sec
*** [pmc: thread 2]   current max clique = 595,  time = 0.183097 sec
*** [pmc: thread 13]   current max clique = 596,  time = 0.28271 sec
-----------------------------------------------------------------------
munmap_chunk(): invalid pointer
Aborted (core dumped)

Could anyone help me to find what is wrong? Thank you so much!

jingnanshi, 2020-05-25:
Can you try `OMP_NUM_THREADS=12 python teaser_python_ply.py` ?

LKeaning, 2020-05-26:
> Can you try `OMP_NUM_THREADS=12 python teaser_python_ply.py` ?
It works! Thank you so much!

closed 2020-05-27 by jingnanshi