MIT-SPARK/TEASER-plusplus #30: teaser_cpp_fpfh

xjsansan, 2020-05-27:
when cmake i got error, do u know why this happen? hoping for your reply.
CMake Error at CMakeLists.txt:12 (add_executable):
  Target "teaser_cpp_fpfh" links to target "teaserpp::teaser_features" but
  the target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?

jingnanshi, 2020-05-27:
I'll see whether I can reproduce this issue. What's your operating system and what's your CMake version?

xjsansan, 2020-05-27:
thx first. i use Cmake 3.3.7 and PCL1.8 in Ubuntu system, it seems i cant link teaser_features. i make BUILD_TEASER_FPFH=ON when cmake

xjsansan, 2020-05-28:
thx first. i use Cmake 3.16.0-rc3, PCL1.8 and Eigen3 3.3.7 in Ubuntu16.04 system, it seems i cant link teaser_features. i make BUILD_TEASER_FPFH=ON when cmake

> I'll see whether I can reproduce this issue. What's your operating system and what's your CMake version?

jingnanshi, 2020-08-12:
Closing this issue because there's no update.

closed 2020-08-12 by jingnanshi