MIT-SPARK/TEASER-plusplus #31: Feeding TEASER-plusplus correspondences

Aimi-no, 2020-06-01:
Hi, when I was reading your paper I found out that correspondences can be fed to TEASER. Is there a way to do it in python? I haven't found any parameters in RobustRegistrationSolver or RobustRegistrationSolver.Params that would indicate the way to do it and I haven't found it in examples.

Thank you.

jingnanshi, 2020-06-01:
Hi,

In python, you need to provide TEASER with `src` and `dst` (two numpy matrices), and these two matrices should have rows corresponding to each other. In other words, you should view `src` and `dst` as corresponding points, not two point clouds with arbitrary arrangements of points.

Aimi-no, 2020-06-01:
@jingnanshi 
I see. Thank you for your answer.

closed 2020-06-01 by Aimi-no

jingnanshi, 2020-06-01:
Np! @Aimi-no it should actually be corresponding columns instead of rows.