MIT-SPARK/TEASER-plusplus #32: how to make the python demo to do registration on more points

LKeaning, 2020-06-04:
I run the demo code:
`import numpy as np
import teaserpp_python

# Generate random data points
src = np.random.rand(3, 10000)

# Apply arbitrary scale, translation and rotation
scale = 1.2
translation = np.array([[1], [0], [-1]])
rotation = np.array([[0.98370992, 0.17903344, -0.01618098],
                     [-0.04165862, 0.13947877, -0.98934839],
                     [-0.17486954, 0.9739059, 0.14466493]])
dst = scale * np.matmul(rotation, src) + translation

# Add two outliers
dst[:, 1] += 10
dst[:, 9] += 15

# Populate the parameters
solver_params = teaserpp_python.RobustRegistrationSolver.Params()
solver_params.cbar2 = 1
solver_params.noise_bound = 0.01
solver_params.estimate_scaling = True 
solver_params.rotation_estimation_algorithm = (
    teaserpp_python.RobustRegistrationSolver.ROTATION_ESTIMATION_ALGORITHM.GNC_TLS
)
solver_params.rotation_gnc_factor = 1.4
solver_params.rotation_max_iterations = 100
solver_params.rotation_cost_threshold = 1e-12
print("TEASER++ Parameters are:", solver_params)
teaserpp_solver = teaserpp_python.RobustRegistrationSolver(solver_params)

solver = teaserpp_python.RobustRegistrationSolver(solver_params)
solver.solve(src, dst)

solution = solver.getSolution()

# Print the solution
print("Solution is:", solution)`

The maximum of number of points in the point cloud I can run successfully is 10000. If the number is greater than 10000, the program will be killed automatically. Is it possible to run the code on more points, since I'm doing face point cloud registration? Thanks!

jingnanshi, 2020-06-04:
Hi @LKeaning, can you check whether you have run out of memory? 

Also, I think you might have a misunderstanding of `src` and `dst`. The `src` and `dst` are not point clouds. They are corresponding points in the source and destination clouds. Thus, it makes more sense for you to have a learned feature detector to generate features and match them, then pass the correspondences to TEASER++.

LKeaning, 2020-06-05:
OK, I got it. Thanks!

closed 2020-06-05 by jingnanshi

idhamari, 2021-02-05:
I have exactly the same problem, I think it would be nice if you add more comments to explain this in the python example.