MIT-SPARK/TEASER-plusplus #33: Installing on mac

negarl, 2020-06-09:
Can I install it on macOS High Sierra?

jingnanshi, 2020-06-09:
@negarl I haven't tested on macOS before unfortunately. Feel free to try & post any questions here!

negarl, 2020-06-13:
Hi again, 
I have too much trouble installing PCL on mac. I got this error, Can you help me with it? 
"clang: error: linker command failed with exit code 1 (use -v to see invocation)"
What if I install PCL with homebrew instead of the one you used ? would it be any different?

jingnanshi, 2020-06-13:
@negarl Hi, the core TEASER++ library does not require PCL. You should try to just compile with BUILD_TEASER_FPFH=OFF

jingnanshi, 2020-08-12:
Closing this issue for now. Feel free to reopen if you have more questions.

closed 2020-08-12 by jingnanshi