MIT-SPARK/TEASER-plusplus #34: teaser_cpp_ply core dumped

evarcurto, 2020-06-15:
I'm trying to run ./teaser_cpp_ply on Ubuntu 16.04 and I got this output: 

`eva-isr@evaisr-Z390-AORUS-ULTRA:~/TEASER-plusplus/examples/teaser_cpp_ply/build$ ./teaser_cpp_ply
	Read 1889 total vertices 
*** [pmc heuristic: thread 1]   current max clique = 505,  time = 0.00122905 sec
*** [pmc heuristic: thread 1]   current max clique = 517,  time = 0.00192809 sec
*** [pmc heuristic: thread 7]   current max clique = 519,  time = 0.0488579 sec
*** [pmc heuristic: thread 9]   current max clique = 520,  time = 0.0491951 sec
*** [pmc heuristic: thread 15]   current max clique = 523,  time = 0.05603 sec
[pmc heuristic]	 mc = 523
Created adjacency matrix in 0.00097394 seconds
[pmc: initial k-core pruning]  before pruning: |V| = 1889, |E| = 332188
[pmc: initial k-core pruning]  after pruning:  |V| = 1889, |E| = 284750
[pmc]  initial pruning took 0.00188708 sec
[pmc: sorting neighbors]  |E| = 569500, |E_sorted| = 569500
|V| = 760
*** [pmc: thread 16]   current max clique = 525,  time = 0.0975699 sec
>>> [pmc: thread 13]
[pmc: bounds updated - thread 13]  time = 0.102387 sec, |V| = 1885 (4 / 1889), |E| = 284750, w = 525, p = 0.159683, d_min = 0, d_avg = 351.708, d_max = 759, k_max = 694
*** [pmc: thread 2]   current max clique = 532,  time = 0.103024 sec
[pmc: bounds updated - thread 16]  time = 0.105693 sec, |V| = 1887 (2 / 1889), |E| = 284750, w = 532, p = 0.159683, d_min = 0, d_avg = 351.708, d_max = 759, k_max = 694
*** [pmc: thread 14]   current max clique = 539,  time = 0.106291 sec
*** [pmc: thread 1]   current max clique = 540,  time = 0.110872 sec
*** [pmc: thread 8]   current max clique = 542,  time = 0.111829 sec
*** [pmc: thread 10]   current max clique = 545,  time = 0.111934 sec
[pmc: bounds updated - thread 14]  time = 0.113517 sec, |V| = 1882 (7 / 1889), |E| = 284750, w = 545, p = 0.159683, d_min = 0, d_avg = 351.708, d_max = 759, k_max = 694
[pmc: bounds updated - thread 15]  time = 0.114689 sec, |V| = 1880 (9 / 1889), |E| = 284750, w = 545, p = 0.159683, d_min = 0, d_avg = 351.708, d_max = 759, k_max = 694
*** [pmc: thread 10]   current max clique = 546,  time = 0.195726 sec
*** [pmc: thread 4]   current max clique = 548,  time = 0.217088 sec
*** [pmc: thread 12]   current max clique = 549,  time = 0.217261 sec
*** [pmc: thread 6]   current max clique = 551,  time = 0.218583 sec
*** [pmc: thread 5]   current max clique = 552,  time = 0.302685 sec
-----------------------------------------------------------------------
*** Error in `./teaser_cpp_ply': free(): invalid next size (fast): 0x000000000187a220 ***
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x777e5)[0x7fa0af4147e5]
/lib/x86_64-linux-gnu/libc.so.6(+0x8037a)[0x7fa0af41d37a]
/lib/x86_64-linux-gnu/libc.so.6(cfree+0x4c)[0x7fa0af42153c]
/usr/local/lib/libpmc.so(_ZN3pmc14pmcx_maxclique12search_denseERNS_9pmc_graphERSt6vectorIiSaIiEE+0xec1)[0x7fa0af182d61]
/usr/local/lib/libteaser_registration.so(_ZN6teaser15MaxCliqueSolver13findMaxCliqueENS_5GraphE+0xf11)[0x7fa0b02890d1]
/usr/local/lib/libteaser_registration.so(_ZN6teaser24RobustRegistrationSolver5solveERKN5Eigen6MatrixIdLi3ELin1ELi0ELi3ELin1EEES5_+0x59a)[0x7fa0b027f06a]
./teaser_cpp_ply[0x4026cf]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf0)[0x7fa0af3bd830]
./teaser_cpp_ply[0x401b19]
======= Memory map: ========
00400000-00428000 r-xp 00000000 103:05 4465302                           /home/eva-isr/TEASER-plusplus/examples/teaser_cpp_ply/build/teaser_cpp_ply
00627000-00628000 r--p 00027000 103:05 4465302                           /home/eva-isr/TEASER-plusplus/examples/teaser_cpp_ply/build/teaser_cpp_ply
00628000-00629000 rw-p 00028000 103:05 4465302                           /home/eva-isr/TEASER-plusplus/examples/teaser_cpp_ply/build/teaser_cpp_ply
01215000-0287f000 rw-p 00000000 00:00 0                                  [heap]
7fa02c000000-7fa02c58e000 rw-p 00000000 00:00 0 
7fa02c58e000-7fa030000000 ---p 00000000 00:00 0 
7fa030000000-7fa030021000 rw-p 00000000 00:00 0 
7fa030021000-7fa034000000 ---p 00000000 00:00 0 
7fa034000000-7fa03431f000 rw-p 00000000 00:00 0 
7fa03431f000-7fa038000000 ---p 00000000 00:00 0 
7fa03c000000-7fa03c372000 rw-p 00000000 00:00 0 
7fa03c372000-7fa040000000 ---p 00000000 00:00 0 
7fa044000000-7fa04436a000 rw-p 00000000 00:00 0 
7fa04436a000-7fa048000000 ---p 00000000 00:00 0 
7fa04c000000-7fa04c36f000 rw-p 00000000 00:00 0 
7fa04c36f000-7fa050000000 ---p 00000000 00:00 0 
7fa054000000-7fa05447b000 rw-p 00000000 00:00 0 
7fa05447b000-7fa058000000 ---p 00000000 00:00 0 
7fa05c000000-7fa05c36a000 rw-p 00000000 00:00 0 
7fa05c36a000-7fa060000000 ---p 00000000 00:00 0 
7fa064000000-7fa06435e000 rw-p 00000000 00:00 0 
7fa06435e000-7fa068000000 ---p 00000000 00:00 0 
7fa06c000000-7fa06c35c000 rw-p 00000000 00:00 0 
7fa06c35c000-7fa070000000 ---p 00000000 00:00 0 
7fa074000000-7fa07435b000 rw-p 00000000 00:00 0 
[... 76 lines cut ...]
7fa0afc88000-7fa0afe5b000 r-xp 00000000 103:04 130844                    /usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.28
7fa0afe5b000-7fa0b005a000 ---p 001d3000 103:04 130844                    /usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.28
7fa0b005a000-7fa0b0065000 r--p 001d2000 103:04 130844                    /usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.28
7fa0b0065000-7fa0b0068000 rw-p 001dd000 103:04 130844                    /usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.28
7fa0b0068000-7fa0b006b000 rw-p 00000000 00:00 0 
7fa0b006b000-7fa0b0071000 r-xp 00000000 103:04 197592                    /usr/local/lib/libteaser_io.so
7fa0b0071000-7fa0b0270000 ---p 00006000 103:04 197592                    /usr/local/lib/libteaser_io.so
7fa0b0270000-7fa0b0271000 r--p 00005000 103:04 197592                    /usr/local/lib/libteaser_io.so
7fa0b0271000-7fa0b0272000 rw-p 00006000 103:04 197592                    /usr/local/lib/libteaser_io.so
7fa0b0272000-7fa0b028c000 r-xp 00000000 103:04 197593                    /usr/local/lib/libteaser_registration.so
7fa0b028c000-7fa0b048b000 ---p 0001a000 103:04 197593                    /usr/local/lib/libteaser_registration.so
7fa0b048b000-7fa0b048c000 r--p 00019000 103:04 197593                    /usr/local/lib/libteaser_registration.so
7fa0b048c000-7fa0b048d000 rw-p 0001a000 103:04 197593                    /usr/local/lib/libteaser_registration.so
7fa0b048d000-7fa0b04b3000 r-xp 00000000 103:04 1575051                   /lib/x86_64-linux-gnu/ld-2.23.so
7fa0b04b5000-7fa0b0671000 rw-p 00000000 00:00 0 
7fa0b06b0000-7fa0b06b2000 rw-p 00000000 00:00 0 
7fa0b06b2000-7fa0b06b3000 r--p 00025000 103:04 1575051                   /lib/x86_64-linux-gnu/ld-2.23.so
7fa0b06b3000-7fa0b06b4000 rw-p 00026000 103:04 1575051                   /lib/x86_64-linux-gnu/ld-2.23.so
7fa0b06b4000-7fa0b06b5000 rw-p 00000000 00:00 0 
7ffec6b39000-7ffec6b5f000 rw-p 00000000 00:00 0                          [stack]
7ffec6b65000-7ffec6b68000 r--p 00000000 00:00 0                          [vvar]
7ffec6b68000-7ffec6b6a000 r-xp 00000000 00:00 0                          [vdso]
ffffffffff600000-ffffffffff601000 r-xp 00000000 00:00 0                  [vsyscall]
Aborted (core dumped)
`

jingnanshi, 2020-06-15:
Try run with OMP_NUM_THREADS=12 (see the bottom of the README)

evarcurto, 2020-06-16:
Unfortunately the output is the same.

```
eva-isr@evaisr-Z390-AORUS-ULTRA:~/TEASER-plusplus/examples/teaser_cpp_ply/build$ ./teaser_cpp_ply OMP_NUM_THREADS=12
	Read 1889 total vertices 
*** [pmc heuristic: thread 1]   current max clique = 519,  time = 0.00533485 sec
*** [pmc heuristic: thread 5]   current max clique = 522,  time = 0.023365 sec
*** [pmc heuristic: thread 3]   current max clique = 524,  time = 0.0473819 sec
*** [pmc heuristic: thread 15]   current max clique = 525,  time = 0.0525689 sec
*** [pmc heuristic: thread 14]   current max clique = 529,  time = 0.0540009 sec
*** [pmc heuristic: thread 6]   current max clique = 531,  time = 0.0564339 sec
*** [pmc heuristic: thread 8]   current max clique = 537,  time = 0.0585339 sec
[pmc heuristic]	 mc = 537
Created adjacency matrix in 0.000977039 seconds
[pmc: initial k-core pruning]  before pruning: |V| = 1889, |E| = 333704
[pmc: initial k-core pruning]  after pruning:  |V| = 1889, |E| = 285091
[pmc]  initial pruning took 0.00185513 sec
[pmc: sorting neighbors]  |E| = 570182, |E_sorted| = 570182
|V| = 760
>>> [pmc: thread 13]
[pmc: bounds updated - thread 13]  time = 0.0139468 sec, |V| = 1888 (1 / 1889), |E| = 285091, w = 537, p = 0.159875, d_min = 0, d_avg = 353.313, d_max = 759, k_max = 697
*** [pmc: thread 7]   current max clique = 543,  time = 0.107095 sec
*** [pmc: thread 6]   current max clique = 544,  time = 0.109052 sec
*** [pmc: thread 16]   current max clique = 547,  time = 0.110105 sec
*** [pmc: thread 2]   current max clique = 552,  time = 0.114848 sec
*** [pmc: thread 5]   current max clique = 553,  time = 0.11669 sec
[pmc: bounds updated - thread 14]  time = 0.11699 sec, |V| = 1877 (12 / 1889), |E| = 285091, w = 553, p = 0.159875, d_min = 0, d_avg = 353.313, d_max = 759, k_max = 697
[pmc: bounds updated - thread 16]  time = 0.117421 sec, |V| = 1883 (6 / 1889), |E| = 285091, w = 553, p = 0.159875, d_min = 0, d_avg = 353.313, d_max = 759, k_max = 697
*** [pmc: thread 3]   current max clique = 558,  time = 0.117549 sec
*** [pmc: thread 15]   current max clique = 562,  time = 0.119494 sec
[pmc: bounds updated - thread 15]  time = 0.121763 sec, |V| = 1875 (14 / 1889), |E| = 285091, w = 562, p = 0.159875, d_min = 0, d_avg = 353.313, d_max = 759, k_max = 697
*** [pmc: thread 8]   current max clique = 563,  time = 0.23013 sec
*** [pmc: thread 5]   current max clique = 564,  time = 0.230269 sec
*** [pmc: thread 11]   current max clique = 566,  time = 0.230346 sec
*** [pmc: thread 9]   current max clique = 567,  time = 0.235713 sec
-----------------------------------------------------------------------
*** Error in `./teaser_cpp_ply': free(): invalid next size (fast): 0x00000000029f2310 ***
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x777e5)[0x7fbd9bba27e5]
/lib/x86_64-linux-gnu/libc.so.6(+0x8037a)[0x7fbd9bbab37a]
/lib/x86_64-linux-gnu/libc.so.6(cfree+0x4c)[0x7fbd9bbaf53c]
/usr/local/lib/libpmc.so(_ZN3pmc14pmcx_maxclique12search_denseERNS_9pmc_graphERSt6vectorIiSaIiEE+0xec1)[0x7fbd9b910d61]
/usr/local/lib/libteaser_registration.so(_ZN6teaser15MaxCliqueSolver13findMaxCliqueENS_5GraphE+0xf11)[0x7fbd9ca170d1]
/usr/local/lib/libteaser_registration.so(_ZN6teaser24RobustRegistrationSolver5solveERKN5Eigen6MatrixIdLi3ELin1ELi0ELi3ELin1EEES5_+0x59a)[0x7fbd9ca0d06a]
./teaser_cpp_ply[0x4026cf]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf0)[0x7fbd9bb4b830]
./teaser_cpp_ply[0x401b19]
======= Memory map: ========
00400000-00428000 r-xp 00000000 103:05 1324581                           /home/eva-isr/TEASER-plusplus/examples/teaser_cpp_ply/build/teaser_cpp_ply
00627000-00628000 r--p 00027000 103:05 1324581                           /home/eva-isr/TEASER-plusplus/examples/teaser_cpp_ply/build/teaser_cpp_ply
00628000-00629000 rw-p 00028000 103:05 1324581                           /home/eva-isr/TEASER-plusplus/examples/teaser_cpp_ply/build/teaser_cpp_ply
0236e000-0380b000 rw-p 00000000 00:00 0                                  [heap]
7fbd1c000000-7fbd1c021000 rw-p 00000000 00:00 0 
7fbd1c021000-7fbd20000000 ---p 00000000 00:00 0 
7fbd24000000-7fbd2437e000 rw-p 00000000 00:00 0 
7fbd2437e000-7fbd28000000 ---p 00000000 00:00 0 
7fbd28000000-7fbd28368000 rw-p 00000000 00:00 0 
7fbd28368000-7fbd2c000000 ---p 00000000 00:00 0 
7fbd2c000000-7fbd2c382000 rw-p 00000000 00:00 0 
7fbd2c382000-7fbd30000000 ---p 00000000 00:00 0 
7fbd34000000-7fbd34474000 rw-p 00000000 00:00 0 
7fbd34474000-7fbd38000000 ---p 00000000 00:00 0 
7fbd3c000000-7fbd3c335000 rw-p 00000000 00:00 0 
7fbd3c335000-7fbd40000000 ---p 00000000 00:00 0 
7fbd44000000-7fbd44357000 rw-p 00000000 00:00 0 
7fbd44357000-7fbd48000000 ---p 00000000 00:00 0 
7fbd4c000000-7fbd4c361000 rw-p 00000000 00:00 0 
7fbd4c361000-7fbd50000000 ---p 00000000 00:00 0 
7fbd54000000-7fbd54373000 rw-p 00000000 00:00 0 
7fbd54373000-7fbd58000000 ---p 00000000 00:00 0 
[... 79 lines cut ...]
7fbd9c416000-7fbd9c5e9000 r-xp 00000000 103:04 130844                    /usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.28
7fbd9c5e9000-7fbd9c7e8000 ---p 001d3000 103:04 130844                    /usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.28
7fbd9c7e8000-7fbd9c7f3000 r--p 001d2000 103:04 130844                    /usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.28
7fbd9c7f3000-7fbd9c7f6000 rw-p 001dd000 103:04 130844                    /usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.28
7fbd9c7f6000-7fbd9c7f9000 rw-p 00000000 00:00 0 
[... cut ...]
7fbd9ce41000-7fbd9ce42000 rw-p 00026000 103:04 1575051                   /lib/x86_64-linux-gnu/ld-2.23.so
7fbd9ce42000-7fbd9ce43000 rw-p 00000000 00:00 0 
7ffe3de08000-7ffe3de2a000 rw-p 00000000 00:00 0                          [stack]
7ffe3df02000-7ffe3df05000 r--p 00000000 00:00 0                          [vvar]
7ffe3df05000-7ffe3df07000 r-xp 00000000 00:00 0                          [vdso]
ffffffffff600000-ffffffffff601000 r-xp 00000000 00:00 0                  [vsyscall]
Aborted (core dumped)
```

jingnanshi, 2020-06-16:
@evarcurto Try this instead:  `OMP_NUM_THREADS=12 ./teaser_cpp_ply`

evarcurto, 2020-06-17:
@jingnanshi It worked! Thank you!

jingnanshi, 2020-06-17:
@evarcurto Glad I can help!

closed 2020-06-17 by jingnanshi