MIT-SPARK/TEASER-plusplus #35: Size limit

rmslick, 2020-06-15:
Is there a limitation to the size of the ply file?  And do you have any methods for reading in .pcd files instead of ply?  Below is the output of teaser_cpp_ply when I attempted with my own ply data.

Read 194507 total vertices 
	Read 164016 total vertices 
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
Aborted (core dumped)

jingnanshi, 2020-06-15:
TEASER++ takes in correspondences, not raw point cloud. So you need to use PCL or some other libraries to first generate features and correspondences. Then feed those correspondences to TEASER++.

rmslick, 2020-06-16:
So, taking  teaser_cpp_ply as an example, you are saying the the ply that is input into the example is a correspondence in the form of a .ply file?

jingnanshi, 2020-06-16:
The example read a .ply file that contains a 3D model, applies random 3D transformation with outliers, and estimated the transformation.

rmslick, 2020-06-16:
Right, so if I input two of my own 3D model represented as a .ply's to the example and I get the error that I listed in my post then I am still curious- is there an upper limit on the size of the ply to this example program?  

I have attempted both to simply swap the example model with my own model and I have loaded two files and took out the random transformation and in both cases I have gotten seg faults and allocation errors.

jingnanshi, 2020-06-16:
> > In your case, because the input and output are of different sizes, Eigen failed to do the proper memory allocation.
> 
> This is the case given the output I posted, but it seg faults if I put in the .ply with 164016 total vertices to the given example and apply the random transformation. In this case there is only one input as is consistent with the example but it still seg faults due to size. (Insofar as I can tell.)

I see. So I don't have a hard memory limit, but 164016 vertices will definitely consume a lot of memory, and seg faulting due to memory will definitely happen if your RAM runs out. You can confirm it by maybe checking with your system monitor.

rmslick, 2020-06-16:
I suspect this is what is happening.  Thank you for your prompt replies!

jingnanshi, 2020-06-16:
@rmslick No problem! Open any new issues if you have new questions.

closed 2020-06-16 by jingnanshi