MIT-SPARK/TEASER-plusplus #37: teaser_cpp_fpfh core dumped

evarcurto, 2020-06-27:
Hi,

I'm trying to run ./teaser_cpp_fpfh.cc on Ubuntu 16.04 and I got this output:

```
OMP_NUM_THREADS=12  ./teaser_cpp_fpfh
	Read 94 total vertices 
CROSS CHECK
Skipping Tuple Constraint.
Segmentation fault (core dumped)
```
I'm using the bun_zipper_res3.ply pointcloud  (after downsampling it).

jingnanshi, 2020-06-27:
Hi @evarcurto, thanks for reporting this error! Do you mind try running the example that does not involve FPFH? For example, can you try running https://github.com/MIT-SPARK/TEASER-plusplus/tree/master/examples/teaser_cpp_ply ?

evarcurto, 2020-06-27:
Hi @jingnanshi ! Thank you for the prompt reply! 
I can now run `teaser_cpp_ply` without any problem. (https://github.com/MIT-SPARK/TEASER-plusplus/issues/34#issuecomment-644445793).
My goal is to use TEASER++ to do the registration of my point clouds dataset but I need to get the correspondences...that's why I'm trying to use the `teaser_cpp_fpfh` example.

jingnanshi, 2020-06-27:
What version of PCL are you using? Also, can you please run all the unit tests and report back failures? Thanks!

evarcurto, 2020-06-28:
I'm using PCL 1.9.1.
After run the tests:

```
69% tests passed, 11 tests failed out of 36

Total Test time (real) =   7.08 sec

The following tests FAILED:
	 15 - RegistrationTest.LargeModel (SEGFAULT)
	 16 - RegistrationTest.SolveForScale (SEGFAULT)
	 17 - RegistrationTest.SolveForRotation (SEGFAULT)
	 18 - RegistrationTest.SolveRegistrationProblemDecoupled (SEGFAULT)
	 19 - RegistrationTest.OutlierDetection (SEGFAULT)
	 20 - RegistrationTest.NoMaxClique (SEGFAULT)
	 21 - RegistrationTest.CliqueFinderModes (SEGFAULT)
	 27 - FPFHTest.CalculateFPFHFeaturesWithPCL (SEGFAULT)
	 28 - FPFHTest.CalculateFPFHFeaturesWithTeaserInterface (SEGFAULT)
	 29 - FPFHMatcherTest.SelfMatching (SEGFAULT)
	 30 - FPFHMatcherTest.MatchCase1 (SEGFAULT)
```

jingnanshi, 2020-06-28:
Make sure you ran the unit tests with `OMP_NUM_THREADS=12`. Meanwhile, can you try to build the library with debug enabled and obtain a stacktrace with GDB?

evarcurto, 2020-06-28:
Yes, I always use OMP_NUM_THREADS=12.
As you suggested, I built teaser++ with -DCMAKE_BUILD_TYPE=Debug.
The output of `OMP_NUM_THREADS=12 ./teaser_cpp_fpfh` was: 

```
	Read 94 total vertices 
CROSS CHECK
Skipping Tuple Constraint.
Starting scale solver.
Scale estimation complete.
Max core number: 0
Num vertices: 3
[pmc heuristic]	 mc = 0
teaser_cpp_fpfh: /home/eva-isr/TEASER-plusplus/teaser/src/graph.cc:93: std::vector<int> teaser::MaxCliqueSolver::findMaxClique(teaser::Graph): Assertion `in.lb != 0' failed.
Aborted (core dumped)

```

jingnanshi, 2020-06-28:
So it looks like the reason is because there is no clique in the graph formed by the correspondences. How did you perform downsampling on the point cloud `bun_zipper_res3.ply`?

evarcurto, 2020-06-29:
I used matlab to do the downsampling (I experimented with the random and gridAverage mode). Meanwhile, I found that if I run without adding noise/outliers it works for your synthetic point cloud (rabbit). 
Unfortunately, when I try to use my point clouds (from different views with little overlap) I don't get the correct transformation.

jingnanshi, 2020-06-29:
@evarcurto I think the reason of the failure is the poor performance of FPFH. I've experienced similar errors also, and it turned out that the FPFH features matching failed to generate > 2 pairs of correct correspondences. If possible, try learned features such as 3DSmoothNet and FCGF.

jingnanshi, 2020-08-12:
Closing this issue now due to no update.

closed 2020-08-12 by jingnanshi