MIT-SPARK/TEASER-plusplus #38: Correspondence Establishing

Pamphlett, 2020-06-29:
Hi,

Thanks for sharing your code!
I'm wondering about the role (or significance) of the putative correspondence that fed to the whole algorithm. 
Why do we need to use a set of more reliable correspondences (e.g. computed by FPFH) instead of some heuristic (or naive) methods (e.g. nearest neighbor)?

As the computation of FPFH consumes a considerable amount of computational resources which will probably hinder the real-time performance of the whole system.

Thx

jingnanshi, 2020-06-29:
Hi @Pamphlett, thank you for your interests! Certainly you can use heuristics to generate correspondences, but if the two point clouds have zero overlap, then it's unlikely to have any inlier correspondences, and the resulted registration solution will be incorrect. 

The correspondence-free case means that all-to-all correspondences are generated and TEASER++ is used to filter out the incorrect ones. This will 1) consume lots of memory 2) be slower than the case if reliable correspondences are fed to TEASER++ instead.

Regarding the computation cost, I agree that FPFH computation is slow. There has been significant progress on learned 3D feature descriptors recently, see e.g. [FCGF](https://github.com/chrischoy/FCGF) and [3DSmoothNet](https://github.com/zgojcic/3DSmoothNet). In particular, FCGF is really fast and might be suitable for real-time applications. In addition, to further reduce computation time, you can downsample the point clouds then compute features.

So to summarize:
- If putative correspondences are somewhat good, it narrows down the solution space TEASER++ needs to consider, thus TEASER++ should run faster.
- Although TEASER++ can work on correspondence-free problems, it will consume a lot of memory and not suitable for real-time applications (unless the point clouds are small).
- Instead of using FPFH, use more recently developed learned feature descriptors instead.

Pamphlett, 2020-06-30:
@jingnanshi I got it! Thanks for your clear answer ;)

jingnanshi, 2020-06-30:
Glad I can help @Pamphlett

closed 2020-06-30 by jingnanshi