MIT-SPARK/TEASER-plusplus #39: Ability to define custom correspondences (Eg. Point to Line, Point to Plane associations)

ameysutavani, 2020-07-05:
Hello,

I was wondering if it is possible to extend TEASER-plusplus to have user-defined correspondences instead of just point-to-point associations?

For instance, CERES requires the user to define a custom CostFunction class which just overloads () operator to get current residues: 
http://ceres-solver.org/nnls_modeling.html

jingnanshi, 2020-07-07:
Hi @ameysutavani, thank you for your interest and suggestion! I took a quick look, and unfortunately I think it will require more work than simply using polymorphism for the residue functions. TEASER++ is formulated for 3D point-to-point registration only, and changing it to accept other correspondences will require some substantial changes if we want to keep the same level of performance. However, if you are interested in learning more about other geometric correspondences, you may find it helpful to take a look at a new paper from our lab: https://arxiv.org/abs/2006.06769.

ameysutavani, 2020-07-07:
Hi @jingnanshi, thanks for the quick response! Yes, I will surely take a look at the paper; thanks for sharing.

closed 2020-07-07 by jingnanshi