MIT-SPARK/TEASER-plusplus #40: Running Teaser++ in ROS

int-smart, 2020-07-18:
Hi

I am trying to build Teaser++ with ROS. I have added the library in the package.xml and CMakeLists.txt find_package call. The teaser package successfully compiles but the package including teaser cannot find teaser. I am getting the following error:
[image]

I am using:
Ubuntu 18.04.4 LTS
ROS Melodic

Thanks

jingnanshi, 2020-07-20:
Hi @int-smart, what version of PCL you are using?

int-smart, 2020-07-20:
Hi. I have PCL version 1.8.1

int-smart, 2020-07-22:
Any updates on this issue.

jingnanshi, 2020-07-22:
@int-smart I haven't been able to take a look, but it's on my todo list.

Pamphlett, 2020-07-23:
@int-smart Try something like `find_package(PCL REQUIRED)` and `include_directories` in the cmakelists.

jingnanshi, 2020-11-21:
@int-smart Hi, you need to install PCL from source. The default PCL that comes with ROS does not play nicely with my current CMake setup.

jingnanshi, 2020-11-22:
Feel free to reopen if you have more questions.

closed 2020-11-22 by jingnanshi