MIT-SPARK/TEASER-plusplus #41: Input and output in different sizes

negarl, 2020-07-20:
Hi. 
When I use TEASER for unmatched src and dst with shapes (3, 7850) and (3, 14571) respectively, I get no out put from the solver.getSolution() just this : 
< Process finished with exit code 139 (interrupted by signal 11: SIGSEGV)> 
Also, in the paper you present experiments where you tell the algorithm to match everything with everything and show it can still give good results because it is very robust to wrong correspondences. So, how is that ? What is my problem ??

jingnanshi, 2020-07-20:
Hi,

Thanks for the interest. You need to generate all-to-all correspondences yourself. TEASER++ won't automatically generate the correspondences for you.

jingnanshi, 2020-07-21:
Close this issue for now. Feel free to reopen if you have more questions.

closed 2020-07-21 by jingnanshi