MIT-SPARK/TEASER-plusplus #42: Noise bound choosing

Pamphlett, 2020-07-23:
Hi,

First, thanks for your insightful work 👍🏼
Well while reading the paper, I have several questions related to the noise bound and hope to get some help here.

- Can I say that the noise bound (which is $\Beta_{i}$ in the paper) in TLS is something like the bound in Trimmed ICP? As both of them try to resist outliers by a 'hard rejection' manner.
- How to properly choose the noise bound in different applications (different scales of the scenarios(int- outdoor), different overlap ratio)?
- As in the experiment section (XI, _E_) stated, we can pass the result of Teaser++ to Algorithm 3 (Optimality Certification) and get the answer whether the result is optimal or not (Teaser++(CERT)).  Is there a way for the algorithm Teaser++ itself to provide an answer to the optimality (or confidence) of its result? And also for the code of Teaser++(CERT), is there any plan to make it open-sourced 😉 ?

Sorry for these messy questions and thanks in advance!

Pamphlett

jingnanshi, 2020-07-23:
@Pamphlett 
1. I took a quick look at the Trimmed ICP paper, and it looks like it is trimming by using only a number of measurements instead of all the measurements. In one sense, it's trying to something similar to TEASER++: rejecting outliers. I didn't find a parameter in the Trimmed ICP paper that corresponds to the noise bound exactly. You can view the noise bound as the threshold above which we reject pairs of measurements. Essentially, the noise bound is a parameter for a robust cost function we call truncated least squares. See this paper for more insights: https://arxiv.org/pdf/1909.08605.pdf 
2. Usually you need to choose according to sensor specification.
3. I'm currently working on the C++ implementation for the certification code. Stay tuned :)

Pamphlett, 2020-07-25:
@jingnanshi 

1. For me, Trimmed ICP uses an implicit bound: by 1. sorting the pair-wise residuals; 2. taking a certain portion of the sorted residual into consideration (the portion parameter can be view as a 'bound' to reject outliers). 
2. Can I say that if I decrease the $\Beta_{i}$ and fix $c_bar$ to one, it makes the algorithm involves fewer correspondences into the objective function? So does $\Beta_{i}$ really related to the scale of the point cloud?
3. Looking forward to the release 😃 !

jingnanshi, 2020-07-25:
@Pamphlett beta relates to the scale because we are filtering based on the calculated scale values based on correspondences.

Will update here if we finish the release. :)

Pamphlett, 2020-07-29:
@jingnanshi 
Hi jingnan,

Seems that in the current release, Geman McClure is used as the weight function rather than the TLS.
See here https://github.com/MIT-SPARK/TEASER-plusplus/blob/2c50d35af14e198deab76cf3dfbe93dddeb8eafd/teaser/src/registration.cc#L253

jingnanshi, 2020-07-29:
@Pamphlett By setting the param struct, you can choose between TLS and GM. See https://github.com/MIT-SPARK/TEASER-plusplus/blob/2c50d35af14e198deab76cf3dfbe93dddeb8eafd/teaser/src/registration.cc#L603 and https://github.com/MIT-SPARK/TEASER-plusplus/blob/2c50d35af14e198deab76cf3dfbe93dddeb8eafd/teaser/include/teaser/registration.h#L719

Pamphlett, 2020-07-29:
Got it! Thx

jingnanshi, 2020-08-12:
#46

closed 2020-08-12 by jingnanshi

negarl, 2020-08-27:
Hi, 
Thanks for your great work. 
Did you finish the certification code? Is it available in python too?

jingnanshi, 2020-08-27:
@negarl Yes it's finished. It's in python as well. I don't have examples / documentation for it yet.

negarl, 2020-08-31:
> @negarl Yes it's finished. It's in python as well. I don't have examples / documentation for it yet.

Oh, great. When the documentation will be ready?

negarl, 2020-12-11:
> @negarl Yes it's finished. It's in python as well. I don't have examples / documentation for it yet.

Hi, I am wondering if you finished the certification documentation ?
thanks