MIT-SPARK/TEASER-plusplus #43: segmentation fault

Enzo-MiMan, 2020-07-24:
Hi,  I tried to use teaser++ to process a series of point cloud (frames). 
namely, register the point cloud in frame k with that in frame k+1,  and so on.

It works whe I only run on one pair of point clouds, but when I implement it in loop, it prompt the error ”zsh: segmentation fault”

My laptop os is MacOS Catalina and I run the executable file with command “OMP_NUM_THREADS=12 ./teaser_cpp_ply”

Is it caused by thread conflict? How can I deal with this?

Thank you.

```
int main() {

  int i,datalen=0;
  std::string num[1770];
  std::ifstream file("/Users/manmi/Documents/data/square_data/mm_data/timestamp.txt");
  while (!file.eof()) 
  file>>num[datalen++];

  
  for (int i = 0; i < 10; ++i) {
    
      // Load the .ply file
      teaser::PLYReader reader;
      teaser::PointCloud src_cloud;
      std::string src_path = "/Users/manmi/Documents/data/square_data/mm_data/mm_ply/" + num[i+3] + ".ply";
      auto status = reader.read(src_path, src_cloud);
      int N = src_cloud.size();

      teaser::PointCloud tgt_cloud;
      std::string tgt_path = "/Users/manmi/Documents/data/square_data/mm_data/mm_ply/" + num[i] + ".ply";
      reader.read(tgt_path, tgt_cloud);
      int M = tgt_cloud.size();

      // Convert the point cloud to Eigen
      Eigen::Matrix<double, 3, Eigen::Dynamic> src(3, N);
      for (size_t i = 0; i < N; ++i) {
        src.col(i) << src_cloud[i].x, src_cloud[i].y, src_cloud[i].z;
      }

      Eigen::Matrix<double, 3, Eigen::Dynamic> tgt(3, M);
      for (size_t i = 0; i < M; ++i) {
        tgt.col(i) << tgt_cloud[i].x, tgt_cloud[i].y, tgt_cloud[i].z;
      }
      
      // Run TEASER++ registration
      // Prepare solver parameters
      teaser::RobustRegistrationSolver::Params params;
      params.noise_bound = NOISE_BOUND;
      params.cbar2 = 1;
      params.estimate_scaling = false;
      params.rotation_max_iterations = 100;
      params.rotation_gnc_factor = 1.4;
      params.rotation_estimation_algorithm =
          teaser::RobustRegistrationSolver::ROTATION_ESTIMATION_ALGORITHM::GNC_TLS;
      params.rotation_cost_threshold = 0.000005;

      // Solve with TEASER++
      teaser::RobustRegistrationSolver solver(params);
      std::chrono::steady_clock::time_point begin = std::chrono::steady_clock::now();
      solver.solve(src, tgt);
      std::chrono::steady_clock::time_point end = std::chrono::steady_clock::now();

      auto solution = solver.getSolution();

      // Compare results
      // std::cout << "=====================================" << std::endl;
      // std::cout << "          TEASER++ Results           " << std::endl;
      // std::cout << "=====================================" << std::endl;
 
      std::ofstream outfile;
      outfile.open("/Users/manmi/Documents/data/square_data/mm_data/mm_T_teaser.txt", std::ios::app);
      outfile << num[i] << " " << solution.translation(0) << " " << solution.translation(1) << " " << solution.translation(2) << std::endl;
      outfile.close();

          
      outfile.open("/Users/manmi/Documents/data/square_data/mm_data/mm_R_matrix_teaser.txt", std::ios::app);
      outfile << num[i] << " "  \
              << solution.rotation(0) << " " << solution.rotation(1) << " " << solution.rotation(2) << " "  \
              << solution.rotation(3) << " " << solution.rotation(4) << " " << solution.rotation(5) << " "  \
              << solution.rotation(6) << " " << solution.rotation(7) << " " << solution.rotation(8) << std::endl;
      outfile.close();
  }

}
```

Output : 

[image]

jingnanshi, 2020-07-24:
@Enzo-MiMan Thank you for your interest. Just to confirm: does N equal to M?

Enzo-MiMan, 2020-07-24:
No, they are different, but all around 30~60 points.

jingnanshi, 2020-07-24:
@Enzo-MiMan They need to be the same size. The reason why it did not segfault before is shear luck. You need to either 1) have a feature extraction algorithm to extract equal number of features from the two point clouds, match them with NN matching, or 2) generate all possible correspondences and feed them directly to TEASER.

Enzo-MiMan, 2020-07-24:
Oh, I see. Thank you so much!

jingnanshi, 2020-07-24:
@Enzo-MiMan No problem. Let me know if you have any more questions!

closed 2020-07-24 by jingnanshi