MIT-SPARK/TEASER-plusplus #44: run error

LAICHAO2018, 2020-08-07:
Thank you for you wonderful work!
I ran into a problem when I ran Mini C + + example according to your steps. I hope you can answer it.

laichao@laichao-PC:~/TEASER-plusplus/examples/teaser_cpp_ply/build$ ./teaser_cpp_ply 
./teaser_cpp_ply: error while loading shared libraries: libpmc.so: cannot open shared object file: No such file or directory

I run in Ubuntu 18.04, Thank you for your response~

LAICHAO2018, 2020-08-07:
I've tried several times and I still can't...

pbeeson, 2020-08-11:
Just add pmc to the target_link_libraries line in the CMake file

qiaozhijian, 2020-10-28:
> Just add pmc to the target_link_libraries line in the CMake file

what's pmc? Could you please give a link?

jingnanshi, 2020-11-23:
@LAICHAO2018 @qiaozhijian You should be able to fix this by running `sudo ldconfig` after installing TEASER++.

idhamari, 2021-02-04:
I had the same error and using 

               sudo ldconfig  

solved it. Probably, it should be added to the installation instruction part.

jingnanshi, 2021-02-04:
@idhamari Thanks for reporting. Will add it to our documentation.

closed 2021-07-17 by jingnanshi