MIT-SPARK/TEASER-plusplus #45: Calculation error

LAICHAO2018, 2020-08-11:
Hello
     Thank you again, When I use example "teaser_cpp_fpfh" to with stanford bunny data, I meet a problem: The error may be too big. Here is my output.

        Read 35947 total vertices
CROSS CHECK
Skipping Tuple Constraint.
*** [pmc heuristic: thread 1]   current max clique = 2,  time = 0 sec
[pmc heuristic]  mc = 2
=====================================
          TEASER++ Results
=====================================
Expected rotation:
  0.996927  0.0668736 -0.0406664
 -0.066129   0.997618  0.0194009
 0.0418676 -0.0166518   0.998978
Estimated rotation:
 0.977325 -0.196892 0.0779098
-0.161481 -0.931042 -0.327237
 0.136968  0.307236 -0.941725
Error (deg): 2.83486

Expected translation:
 -0.115577
-0.0387705
  0.114875
Estimated translation:
-179.815
 57.6956
 41.1567
Error (m): 193.157

Number of correspondences: 35947
Number of outliers: 1700
Time taken (s): 0.007342

jingnanshi, 2020-08-11:
Hi @LAICHAO2018, thank you for your interest. It looks like the maximum clique is too small. I'll take a look at the example.

LAICHAO2018, 2020-08-12:
> Hi @LAICHAO2018, thank you for your interest. It looks like the maximum clique is too small. I'll take a look at the example.

maybe you can try this dataset，I have been tested in teaser_cpp_fpfh and teaser_cpp_ply, The result error is a little big... 
[https://github.com/LAICHAO2018/bunny_datasets](url)

jingnanshi, 2020-08-12:
Hi @LAICHAO2018,

1. The examples you provided are of different scale, and I don't think FPFH can handle objects of different scale well. 
2. If you didn't change the parameter to estimate scale, then the provided code examples are not going to work.

LAICHAO2018, 2020-08-12:
I modified "estimate_scaling = True"

LAICHAO2018, 2020-08-12:
I modified "estimate_scaling = true"

jingnanshi, 2020-08-12:
Hi @LAICHAO2018, then the issue is probably caused by FPFH features. FPFH features use a radius to estimate the local curvature, which is not going to work with objects of different scales. You can adjust the radius, but that will require you to know the scale beforehand.

closed 2020-11-21 by jingnanshi