MIT-SPARK/TEASER-plusplus #47: Information about finding multiple solutions

pbeeson, 2020-08-11:
First, thanks for providing such well organized source code that's been pretty straightforward to install and get working with my own data.  

I was wondering if it is possible to get multiple transform solutions out using Teaser.   In particular, I was investigating using Teaser for 6D pose estimation of objects in depth data.  I have very noisy scene data and very clean (CAD) model info.  I have something working in a commercial software package, and pretty quickly got a similar result from Teaser to find the correct object pose in a cluttered scene.

However, most of the time, there are going to be multiple instances of the same object type in the scene, so I'd like to find all matches.  I supposed one way I could do this is to take the first solution, find all points in the scene that are closest points to hte object at the solution, and remove those points from the scene (presumably, I'd need to remove the appropriate FPFH features and their correspondences too -- or recompute them assuming that's not to expensive), then re-run the Teaser solver; however, this seems like a lot of work if there's already a simpler solution.  This also assumes that the FPFH correspondence matcher that you wrote returns global correspondences for the source and target, and hasn't culled them down to some base set used for the solver.

Any help you might provide would be appreciated.  Thanks in advance.

jingnanshi, 2020-08-11:
Hi @pbeeson, thank you for your interest and compliments :)

> I was wondering if it is possible to get multiple transform solutions out using Teaser. 

Currently, TEASER++ assumes there's only one solution unfortunately.

> However, most of the time, there are going to be multiple instances of the same object type in the scene, so I'd like to find all matches. I supposed one way I could do this is to take the first [...]

That's my first thought as well. I think my FPFH matcher currently returns global correspondences, but I need to double check. My question is: do you know the number of instances in the scene? 

If not, the only elegant solution I can think of is to have a classification/segmentation network to first obtain 3D segmentation, and then doing some sort of clustering to obtain different clusters of 3D points to run TEASER++ on.

If you do know the number of instances in the scene, then you can try the method you just described. Alternatively, it is possible to pick the top N cliques, where N is the number of instances, and run TEASER++ on each of the clique. 

On a side note, I'm suprised that FPFH matcher is working. My experience has been poor with FPFH, and have to use some sort of learned descriptors for feature matching such as https://github.com/chrischoy/FCGF. 

Considering multiple hypotheses robustly is a research direction that we are currently working on. Let me know if you have more questions.

pbeeson, 2020-08-11:
FPFH isn't working that great after all.  It got the translation close but when I finally visualized the transformed object, the rotation is poor.

pbeeson, 2020-08-11:
Unfrotunately, for many application, learning simply isn't an option.  I was looking to use SHOT or USC features instead of FPFH, but it looks like you have a very special set of matching criteria in Teaser that is different than just finding the one nearest neighbor for each source feature in the target.  So, it'll be some work to make that a bit more generic to handle any kind of PCL descriptor.

jingnanshi, 2020-08-12:
@pbeeson The creteria are quite arbitrary --- those used in the FPFH matching code here are not by any means the best. For the experiments mentioned in our paper, we simply used **bi-directional nearest neighbor**: (p1,p2), where p1 is a src point and p2 is a dst point, is used as a correspondence if and only if p1's feature is the closest to p2's feature among all points in src and p2's feature is the closest to p1's feature among all dst points.

jingnanshi, 2020-08-19:
@pbeeson Closing this issue for now. Feel free to open other issues if you have more questions.

closed 2020-08-19 by jingnanshi

agroenenberg, 2021-02-09:
> FPFH isn't working that great after all. It got the translation close but when I finally visualized the transformed object, the rotation is poor.

Hi, any updates on how to improve this? I have the exact same problem, the position is found most of the time, however, the rotation estimation is poor. Learning isn't an option for my application either. Is there any feature extraction tool that you found that worked best for a CAD model to cluttered scene registration? Thanks!

agroenenberg, 2021-02-10:
> @agroenenberg You can try a few things. First, compute the number of inlier correspondences vs. outlier correspondences. If the ratio of inlier correspondences is less than 1%, then it's hard to have accurate pose estimation. In addition, try adding ICP right after TEASER to see whether you can improve the rotation estimates.

Thanks for the quick reply, the fpfh descriptor generated 796 correspondences from which 238 were considered inliers according to `len(solver.getTranslationInliersMap())`. Still, no proper rotation. I'm already using ICP after Teaser.