MIT-SPARK/TEASER-plusplus #50: Minimum Eigen version

YuePanEdward, 2020-08-16:
Thank you very much for this terrific open-source project. It works quite well in my LiDAR SLAM project even with a 2% correspondence inlier rate. 

I found that there are some operator overloading error when building with Eigen 3.2. However, the building passed with Eigen version >= 3.3. So maybe the min required Eigen version should be 3.3 instead of 3.2 in CMakeList.

jingnanshi, 2020-08-17:
@YuePanEdward thank you for the kind words and feedback! Do you mind posting the errors here so I can take a look? I originally set it to 3.3, but someone suggested setting it to 3.2 so Ubuntu 16.04's default Eigen will work.

YuePanEdward, 2020-08-17:
### errors with Ubuntu 16.04's default Eigen (3.2.9)
 ```
TEASER-plusplus-2.0/teaser/src/certification.cc:100:46: error: no match for ‘operator-’ (operand types are ‘Eigen::MatrixXd {aka Eigen::Matrix<double, -1, -1>}’ and ‘const ScalarMultipleReturnType {aka const Eigen::CwiseUnaryOp<Eigen::internal::scalar_multiple_op<double>, const Eigen::SparseMatrix<double> >}’)
   Eigen::SparseMatrix<double> M_init = Q_bar - mu * J_bar - lambda_bar_init;
```
 ```
TEASER-plusplus-2.0/teaser/src/certification.cc:133:25: error: no match for ‘operator-’ (operand types are ‘const Eigen::CwiseBinaryOp<Eigen::internal::scalar_difference_op<double>, const Eigen::CwiseUnaryOp<Eigen::internal::scalar_multiple_op<double>, const Eigen::Matrix<double, -1, -1> >, const Eigen::Matrix<double, -1, -1> >’ and ‘Eigen::SparseMatrix<double>’)
     temp_W = 2 * M_PSD-M-M_init;
```
```
TEASER-plusplus-2.0/teaser/src/certification.cc:140:22: error: no match for ‘operator+’ (operand types are ‘Eigen::SparseMatrix<double>’ and ‘Eigen::MatrixXd {aka Eigen::Matrix<double, -1, -1>}’)
     M_affine = M_init+W_dual;
```

Maybe this is my own mistake.

jingnanshi, 2020-08-18:
@YuePanEdward thank you. It's not your mistake. This is caused by the new certification module I added. Will fix soon.

YuePanEdward, 2020-08-18:
Thank you a lot.
By the way, some examples of the usage of certification API would be helpful.

jingnanshi, 2020-11-16:
Related to this: https://eigen.tuxfamily.org/bz/show_bug.cgi?id=632

closed 2020-11-16 by jingnanshi