MIT-SPARK/TEASER-plusplus #51: performance about velodyne sparse point cloud

narutojxl, 2020-08-19:
Hi @jingnanshi, thanks for your sharing to the community!
I found the example data is dense point cloud something to scene reconstruction. Could this project suit for velodyne-16 sparse point cloud to registration, where collected from the outdoor unstructured environment?
Thanks for your help and time !

jingnanshi, 2020-08-19:
@narutojxl Thank you for your interest. Essentially TEASER++ is solving an optimization problem involving 3xN matrices that represent correspondences. It doesn't matter whether the underlying point clouds are indoor vs. outdoor, sparse vs. dense.

That being said, I want to make sure that you understand TEASER++ needs correspondences as inputs. So if you have a feature descriptor that can give you good enough correspondences on LIDAR scans, then TEASER++ will certainly work.

narutojxl, 2020-08-19:
@jingnanshi,thanks very much!  :)

closed 2020-08-19 by narutojxl