MIT-SPARK/TEASER-plusplus #53: When I run teaser_cpp_fpfh example,I run into this problem

jycc1991, 2020-08-20:
When I run teaser_cpp_fpfh example,I run into this problem:
	Read 1889 total vertices 
CROSS CHECK
Skipping Tuple Constraint.
Starting scale solver.
Scale estimation complete.
Max core number: 4
Num vertices: 7
*** [pmc heuristic: thread 1]   current max clique = 5,  time = 1.90735e-05 sec
[pmc heuristic]	 mc = 5
Max Clique of scale estimation inliers: 
Using chain graph for GNC rotation.
Starting rotation solver.
GNC rotation estimation noise bound:0.1
GNC rotation estimation noise bound squared:0.01
GNC-TLS terminated because maximum residual at initialization is very small.
Rotation estimation complete.
Starting translation solver.
Translation estimation complete.
double free or corruption (out)
Aborted (core dumped)

jingnanshi, 2020-08-20:
Please report the exact commands you have used. Also, take a look at this page: https://teaser.readthedocs.io/en/latest/known_issues.html

jycc1991, 2020-08-20:
> Please report the exact commands you have used. Also, take a look at this page: https://teaser.readthedocs.io/en/latest/known_issues.html

OMP_NUM_THREADS=4 ./teaser_cpp_fpfh

jingnanshi, 2020-08-20:
@jycc1991 Compile in debug and post the backtrace here. Also, make sure you have compiled and install PCL from source.

hmz-15, 2020-08-24:
I found that the error can be avoided by comment the line "translation_inliers_ = utils::findNonzero<bool>(translation_inliers_mask_);" after "TEASER_DEBUG_INFO_MSG("Translation estimation complete.");"

jingnanshi, 2020-08-24:
@hmz-15 Thanks for the feedback. I'll take a look.

jycc1991, 2020-08-25:
> @jycc1991 Compile in debug and post the backtrace here. Also, make sure you have compiled and install PCL from source.

It exetutes all well in debug mode.

willat343, 2020-09-01:
I have the same issue (kinda).

If the example is built in debug mode using `-DCMAKE_BUILD_TYPE=Debug`, the output is variable. Sometimes it is the same as in @jycc1991's original issue, sometimes it runs to completion without Seg fault, and sometimes it has even less output before Seg faulting.

If the example is build in release mode using `-DCMAKE_BUILD_TYPE=Release` it consistently runs with a seg fault at the very end:
```
	Read 1889 total vertices 
CROSS CHECK
Skipping Tuple Constraint.
Starting scale solver.
Scale estimation complete.
Max core number: 2
Num vertices: 4
*** [pmc heuristic: thread 2]   current max clique = 3,  time = 2.31266e-05 sec
[pmc heuristic]	 mc = 3
Max Clique of scale estimation inliers: 
Using chain graph for GNC rotation.
Starting rotation solver.
GNC rotation estimation noise bound:0.1
GNC rotation estimation noise bound squared:0.01
GNC-TLS terminated because maximum residual at initialization is very small.
Rotation estimation complete.
Starting translation solver.
Translation estimation complete.
=====================================
          TEASER++ Results           
=====================================
Expected rotation: 
  0.996927  0.0668736 -0.0406664
 -0.066129   0.997618  0.0194009
 0.0418676 -0.0166518   0.998978
Estimated rotation: 
 0.0782653  -0.986769   0.141993
  0.407279    0.16165   0.898885
 -0.909944 -0.0125209   0.414542
Error (deg): 1.80582

Expected translation: 
 -0.115577
-0.0387705
  0.114875
Estimated translation: 
-0.0759707
 0.0511562
 0.0496321
Error (m): 0.11795

Number of correspondences: 1889
Number of outliers: 1700
Time taken (s): 0.002319
Segmentation fault (core dumped)
```

I have installed PCL v1.9.1 from source. I also strangly in the `CMakeLists.txt` had to remove the `teaserpp::` namespace in front of `teaserpp::teaser_features` in order for my linker to find the library despite installing teaserpp with the FPFH option ON and running `sudo ldconfig` afterwards - seems like a separate cmake config issue.

jingnanshi, 2020-09-01:
@willat343 Thanks for the report. My guess is my latest update might have caused this issue. I'll try to take a look soon.

willat343, 2020-09-08:
hi @jingnanshi and @tpatten, has there been any progress on this bug?

tpatten, 2020-09-08:
No, I just removed the line pointed to by @hmz-15 but my code still has the same error as you reported most recently. I haven't had time to investigate further. I am hoping @jingnanshi can help out.

narutojxl, 2020-10-27:
Hi @jingnanshi 
Similar problem.
Build `teaser_cpp_fpfh` release and debug type, when run `OMP_NUM_THREADS=4 ./teaser_cpp_fpfh`,  the segment fault still exist.
* build teaser with PCL 1.9.1, BUILD_TEASER_FPFH = ON, not set build type(use the default setting: Release)
* `teaser_cpp_ply` example is OK.
* when gdb `teaser_cpp_fpfh`, the output is:
>
	(gdb) run 
	Starting program: /home/jxl/Documents/study/teaser++_learn/examples/teaser_cpp_fpfh/build/teaser_cpp_fpfh 
	[Thread debugging using libthread_db enabled]
	Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
		Read 1889 total vertices 
	CROSS CHECK
	Skipping Tuple Constraint.
	[New Thread 0x7fffefa3e700 (LWP 10866)]
	[New Thread 0x7fffef23d700 (LWP 10867)]
	[New Thread 0x7fffeea3c700 (LWP 10868)]
	[New Thread 0x7fffee23b700 (LWP 10869)]
	[New Thread 0x7fffeda3a700 (LWP 10870)]
	[New Thread 0x7fffed239700 (LWP 10871)]
	[New Thread 0x7fffeca38700 (LWP 10872)]
	[New Thread 0x7fffec237700 (LWP 10873)]
	[New Thread 0x7fffeba36700 (LWP 10874)]
	[New Thread 0x7fffeb235700 (LWP 10875)]
	[New Thread 0x7fffeaa34700 (LWP 10876)]
	Starting scale solver.
	Scale estimation complete.
	Max core number: 1
	Num vertices: 3
	*** [pmc heuristic: thread 1]   current max clique = 2,  time = 2.09808e-05 sec
	[pmc heuristic]	 mc = 2
	Max Clique of scale estimation inliers: 
	Using chain graph for GNC rotation.
	Starting rotation solver.
	GNC rotation estimation noise bound:0.1
	GNC rotation estimation noise bound squared:0.01
	GNC-TLS terminated because maximum residual at initialization is very small.
	Rotation estimation complete.
	Starting translation solver.
	Translation estimation complete.
	*** Error in `/home/jxl/Documents/study/teaser++_learn/examples/teaser_cpp_fpfh/build/teaser_cpp_fpfh': double free or corruption (out): 0x00000000006df320 ***
	======= Backtrace: =========
	/lib/x86_64-linux-gnu/libc.so.6(+0x777e5)[0x7ffff3c647e5]
	/lib/x86_64-linux-gnu/libc.so.6(+0x8037a)[0x7ffff3c6d37a]
	/lib/x86_64-linux-gnu/libc.so.6(cfree+0x4c)[0x7ffff3c7153c]
	/usr/local/lib/libteaser_registration.so(_ZN6teaser24RobustRegistrationSolver5solveERKN5Eigen6MatrixIdLi3ELin1ELi0ELi3ELin1EEES5_+0x1156)[0x7ffff7bc6e96]
	/usr/local/lib/libteaser_registration.so(_ZN6teaser24RobustRegistrationSolver5solveERKNS_10PointCloudES3_St6vectorISt4pairIiiESaIS6_EE+0x1b7)[0x7ffff7bc76a7]
	/home/jxl/Documents/study/teaser++_learn/examples/teaser_cpp_fpfh/build/teaser_cpp_fpfh[0x407563]
	/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf0)[0x7ffff3c0d830]
	/home/jxl/Documents/study/teaser++_learn/examples/teaser_cpp_fpfh/build/teaser_cpp_fpfh[0x406779]
backtrace is:
>
	Thread 1 "teaser_cpp_fpfh" received signal SIGABRT, Aborted.
	0x00007ffff3c22428 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:54
	54	../sysdeps/unix/sysv/linux/raise.c: 没有那个文件或目录.
	(gdb) bt
	#0  0x00007ffff3c22428 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:54
	#1  0x00007ffff3c2402a in __GI_abort () at abort.c:89
	#2  0x00007ffff3c647ea in __libc_message (do_abort=do_abort@entry=2, fmt=fmt@entry=0x7ffff3d7ded8 "*** Error in `%s': %s: 0x%s ***\n") at ../sysdeps/posix/libc_fatal.c:175
	#3  0x00007ffff3c6d37a in malloc_printerr (ar_ptr=<optimized out>, ptr=<optimized out>, str=0x7ffff3d7dfe8 "double free or corruption (out)", action=3) at malloc.c:5006
	#4  _int_free (av=<optimized out>, p=<optimized out>, have_lock=0) at malloc.c:3867
	#5  0x00007ffff3c7153c in __GI___libc_free (mem=<optimized out>) at malloc.c:2968
	#6  0x00007ffff7bc6e96 in teaser::RobustRegistrationSolver::solve(Eigen::Matrix<double, 3, -1, 0, 3, -1> const&, Eigen::Matrix<double, 3, -1, 0, 3, -1> const&) ()
	   from /usr/local/lib/libteaser_registration.so
	#7  0x00007ffff7bc76a7 in teaser::RobustRegistrationSolver::solve(teaser::PointCloud const&, teaser::PointCloud const&, std::vector<std::pair<int, int>, std::allocator<std::pair<int, int> > >) () from /usr/local/lib/libteaser_registration.so
	#8  0x0000000000407563 in main () at /home/jxl/Documents/study/teaser++_learn/examples/teaser_cpp_fpfh/teaser_cpp_fpfh.cc:114
	(gdb) 

teaser_cpp_fpfh.cc:114
`solver.solve(src_cloud, tgt_cloud, correspondences); `

Thanks for your help!

jingnanshi, 2020-10-27:
@narutojxl Hi, thanks for reporting the bug. I won't have time to take a look until next week.

narutojxl, 2020-10-28:
It's OK :), thanks in advance!

jingnanshi, 2020-11-23:
@narutojxl Hi, can you try with -DBUILD_WITH_MARCH_NATIVE=ON for TEASER++? PCL default uses -march=native, which causes seg faults in the FPFH example when I tried on my machine.

jingnanshi, 2020-11-23:
@willat343 @tpatten Can you also try building TEASER++ with -DBUILD_WITH_MARCH_NATIVE=ON? Thanks

willat343, 2021-03-13:
hi @jingnanshi, my apologies for the long delay. Good news! After reinstalling (even in Release, which was very problematic before) and running the example several times with `./teaser_cpp_fpfh` and `OMP_NUM_THREADS=4 ./teaser_cpp_fpfh`, I experience no seg faults or unexpected behaviour.

I buiilt/installed teaser with:
```
mkdir build && cd build
cmake -DCMAKE_BUILD_TYPE=Release -DBUILD_TEASER_FPFH=ON -DBUILD_WITH_MARCH_NATIVE=ON ..
make -j$(nproc)
sudo make install
sudo ldconfig
```

This could be a good opportunity for some CMake magic. The BUILD_WITH_MARCH_NATIVE flag could be turned ON automatically if the user compiles teaser with the problematic PCL flags, like BUILD_TEASER_FPFH. I'm not sure if this would have undesirable side effects, since to my knowledge march native can lead to non-portable binaries/libraries. An alternative could be to print a CMake warning if BUILD_TEASER_FPFH is on and BUILD_WITH_MARCH_NATIVE isn't.

I am yet to see if I run into any fpfh issues using Teaser in my own projects, but I will report back to this thread if so.

jingnanshi, 2021-03-13:
@willat343 Thanks for reporting. I'll add a note in the documentation and print out some warning messages in CMake.

zzodo, 2021-03-19:
Hi, I still have an issue with my program using TEASER++.
I have tried all the methods described in earlier comments, but a segmentation fault error occurs when calculating FPFH features with TEASER++ wrapper.
The exact code line is:
`auto descriptors1 = fpfh.computeFPFHFeatures(teaser_pcd1, 0.30, 0.50);`
This behavior only occurred in release mode, not debug mode.

Any solutions?

jingnanshi, 2021-03-19:
Hi @zzodo, it looks like there might be some undefined behaviors in the code. For a temporary fix, you should instead compute FPFH features using PCL directly and pass the matched correspondences to TEASER++.

jingnanshi, 2021-07-17:
Addressed in PR #97 and #107

closed 2021-07-17 by jingnanshi