MIT-SPARK/TEASER-plusplus #54: [QUESTION] Segmentation fault (core dumped)

powerm, 2020-09-10:
**Have you read the documentation?**

- [ ] Yes

Post your theoretical questions / usage questions here.
I have run the example, but  I use my point cloud, but  I have get the result blew:

==================================================
        TEASER++ Python registration example      
==================================================
Starting scale solver.
Segmentation fault (core dumped)

So I don't know what wrong with my  code ?

    solver_params = teaserpp_python.RobustRegistrationSolver.Params()
    solver_params.cbar2 = 1
    solver_params.noise_bound = NOISE_BOUND
    solver_params.estimate_scaling = False
    solver_params.inlier_selection_mode = teaserpp_python.RobustRegistrationSolver.INLIER_SELECTION_MODE.PMC_HEU
    solver_params.rotation_estimation_algorithm = teaserpp_python.RobustRegistrationSolver.ROTATION_ESTIMATION_ALGORITHM.GNC_TLS
    solver_params.rotation_gnc_factor = 1.4
    solver_params.rotation_max_iterations = 100
    solver_params.rotation_cost_threshold = 1e-12
    solver_params.kcore_heuristic_threshold = 0.3

above is my solver params

jingnanshi, 2020-11-21:
@powerm Can you share the arguments you supplied to CMake for building TEASER++? Did you enable BUILD_TEASER_FPFH?

Liys0558, 2021-01-28:
I have the same problem. How to solve?

jingnanshi, 2021-01-28:
@Liys0558 please open a new issue with the exact steps you used to build TEASER++ & run the example. Thank you.

chris9182, 2021-04-29:
I am was having similar issues and resolved this by calling the executable with the prefix `OMP_NUM_THREADS=12`, example: `OMP_NUM_THREADS=12 ./teaser_cpp_ply` .
Is there any way to make this work with more than 12 threads?

jingnanshi, 2021-04-29:
@chris9182 The bug is in the PMC library we are currently using for max clique finding. I'm still working on a fix (and by fix I mean actually locating the bug :( ), but currently OMP_NUM_THREADS=12 is the only solution.

b15h0y, 2021-07-09:
I am having the same issue while running the python example. Is there is a way to fix it for python ? the cpp version works in my case without setting `OMP_NUM_THREADS`

fdila, 2022-03-17:
Does anyone know a workaround for the python binding?
I have tried setting ```os.environ['OMP_NUM_THREADS'] = "12"``` but I still run into segmentation faults.

kaitaotang, 2022-05-16:
> 

this reply works well for me. Thank you

kaitaotang, 2022-05-16:
> I am was having similar issues and resolved this by calling the executable with the prefix `OMP_NUM_THREADS=12`, example: `OMP_NUM_THREADS=12 ./teaser_cpp_ply` . Is there any way to make this work with more than 12 threads?

this reply works well for me. Thank you

gitouni, 2022-06-15:
> 1. INLIER_SELECTION_MODE

Actually, setting `OMP_NUM_THREADS` through `os.environ()` function does not work. May be the reason is PMC is built through cpp that binds to python and `os.environ` dosen't pass this environment parameter to PMC. The solution is that:

```bash
OMP_NUM_THREADS=12 python xxx.py
```

jingnanshi, 2022-07-08:
@fdila sorry for the late response, but I have fixed the seg fault issue caused by mis-allocating lists by using the incorrect number of threads in my fork of the PMC library. In addition, after each solve, you need to call `reset()` on the solver before running again.

closed 2023-05-22 by jingnanshi