MIT-SPARK/TEASER-plusplus #55: SIGABRT During rotation solver

RiccardoGiubilato, 2020-09-15:
I'd like to report an issue I am having on user-supplied pointclouds. 
While estimating the rotation TEASER randomly stops after a failed internal check for Eigen.
The issue usually happens while trying to match **non-matching** clouds, therefore purely wrong keypoint correspondences.
Using Eigen 3.3.2 

Relevant code:
```
 teaser::RobustRegistrationSolver::Params params;
 params.noise_bound = 0.05;
 params.cbar2 = 1;
 params.estimate_scaling = false;
 params.rotation_max_iterations = 100;
 params.rotation_gnc_factor = 1.4;
 params.rotation_estimation_algorithm =
	 teaser::RobustRegistrationSolver::ROTATION_ESTIMATION_ALGORITHM::GNC_TLS;

 // Solve with TEASER++
 teaser::RobustRegistrationSolver solver(params);
 solver.solve(cloud1, cloud0, correspondences);
```

Terminal output: 
```
Starting scale solver.
Scale estimation complete.
Max core number: 7
Num vertices: 138
*** [pmc heuristic: thread 7]   current max clique = 5,  time = 0.00120902 sec
[pmc heuristic]	 mc = 5
Created adjacency matrix in 0.000298023 seconds
[pmc: initial k-core pruning]  before pruning: |V| = 137, |E| = 424
[pmc: initial k-core pruning]  after pruning:  |V| = 79, |E| = 233
[pmc]  initial pruning took 9.67979e-05 sec
[pmc: sorting neighbors]  |E| = 466, |E_sorted| = 466
|V| = 51
-----------------------------------------------------------------------
Max Clique of scale estimation inliers: 
101 103 106 111 133 
Using chain graph for GNC rotation.
Starting rotation solver.
GNC rotation estimation noise bound:0.1
GNC rotation estimation noise bound squared:0.01
matcher_node: /home/libs/Eigen/3.3.2/include/Eigen/src/SVD/SVDBase.h:85: const MatrixUType& Eigen::SVDBase<Derived>::matrixU() const [with Derived = Eigen::JacobiSVD<Eigen::Matrix<double, 3, 3>, 2>; Eigen::SVDBase<Derived>::MatrixUType = Eigen::Matrix<double, 3, 3>; typename Eigen::internal::traits<T>::MatrixType::Scalar = double]: Assertion `m_isInitialized && "SVD is not initialized."' failed.

process with pid 9654 terminated by signal SIGABRT with status: 134 (core dumped!)
```

jingnanshi, 2020-09-15:
Thank you for reporting this issue @RiccardoGiubilato. I don't have much time recently, but I will check the code once my schedule frees up.

RiccardoGiubilato, 2020-09-16:
Thank you @jingnanshi

jingnanshi, 2020-11-21:
@RiccardoGiubilato Do you mind share a set of data that causes this bug? Thanks.

jingnanshi, 2021-07-17:
Closing due to no response. Feel free to reopen if you still have this issue.

hiyyg, 2022-10-13:
Have the same issue. How to solve this problem? @RiccardoGiubilato

jingnanshi, 2022-10-13:
> Have the same issue. How to solve this problem? @RiccardoGiubilato

@hiyyg Do you mind pose all the error messages? Also if possible, please share the data.

hiyyg, 2022-10-13:
@jingnanshi I faced with this bug with a different program. But I think the root cause should be the same.

LimHyungTae, 2024-04-03:
First of all, do you guys run it as follows?
```
OMP_NUM_THREADS=8 ./${YOUR EXACUTABLE FILENAME}
```
8 is just an arbitrary number, so please adjust it.

LimHyungTae, 2024-05-02:
If there are no further questions, we will close the issue :) Thx

closed 2024-05-02 by jingnanshi