MIT-SPARK/TEASER-plusplus #56: Uncompitble with PCL-1.11.1 [BUG]

cdb0y511, 2020-09-28:
**Describe the bug**
The example "teaser_cpp_ply.cc" works fine.
However, if I  

target_link_libraries(${targetname}
        Eigen3::Eigen
        /usr/local/lib/libteaser_io.so
        /usr/local/lib/libteaser_registration.so
        ${PCL_LIBRARIES}
        )

The compile works, but Signal: SIGSEGV (Segmentation fault) comes out. It traces back to Eigen's memory.h

double free or corruption (out)

/** \internal Frees memory allocated with handmade_aligned_malloc */
inline void handmade_aligned_free(void *ptr)
{
  if (ptr) std::free(*(reinterpret_cast<void**>(ptr) - 1));
}

I use the latest PCL 1.11.1.  It works with my project.  So could you update your code for the latest PCL 1.11.1?
Thanks for your great work.

**Have you run the unit tests?**

- [x ] Yes
- [] No --- then this issue will be closed.

**Installed Dependencies**
ubuntu 18.04
PCL1.11.1

**To Reproduce**
Please post the exact steps you executed to reproduce the bug.

**Additional context**
Add any other context about the problem here.

jingnanshi, 2020-09-28:
Hi, thank you for reporting the bug @cdb0y511. I won't have free time to investigate this unfortunately. Feel free to dive in the code and make a PR if you can.

cdb0y511, 2020-09-28:
Thanks for the quick reply, If I have any progress I will let you know. For now, I will use a standalone version of teaserpp for the test.

cdb0y511, 2020-11-28:
It still bothers me. When I enable -march = native, it works for some scene reconstruction. However, it gets a segment fault occasionally. I think it relates to Eigen alignment; I doubt the alignment of PCL 1.11.1 is different from  Teaser. So when they are linked together, memory fault happens occasionally.
omp_set_threads is set to 12 before running.

jingnanshi, 2020-11-28:
Can you try comment out https://github.com/MIT-SPARK/TEASER-plusplus/blob/159726a9587d40b6bb7ece78c1a8362d396a5f2f/teaser/include/teaser/registration.h#L36?

cdb0y511, 2020-11-29:
> Can you try comment out
> 
> https://github.com/MIT-SPARK/TEASER-plusplus/blob/159726a9587d40b6bb7ece78c1a8362d396a5f2f/teaser/include/teaser/registration.h#L36
> 
> ?
No, it doesn't resolve the problem.
For more details
it fails like:

   free(): invalid next size (normal)
or 
   Signal: SIGSEGV (Segmentation fault)

and usually, traces back to 

registration.cc
 
   max_clique_ = clique_solver.findMaxClique(inlier_graph_);

graph.cc
  
    in.lb = maxclique.search(G, C);

and 

  pmc::pmc_heu::serach_bounds(pmc::pmc_graph&, std::vector<int, std::allocator>>&)

cdb0y511, 2020-11-29:
or after solver.reset(param)
trace back to 

  __GI_raise 0x00007f32dd4d8fb7
__GI_abort 0x00007f32dd4da921
__libc_message 0x00007f32dd523967
malloc_printerr 0x00007f32dd52a9da
_int_free 0x00007f32dd53219d
__GI___libc_free 0x00007f32dd53219d
__gnu_cxx::new_allocator<int>::deallocate new_allocator.h:125
std::allocator_traits<std::allocator<int> >::deallocate alloc_traits.h:462
std::_Vector_base<int, std::allocator<int> >::_M_deallocate stl_vector.h:180
std::_Vector_base<int, std::allocator<int> >::~_Vector_base stl_vector.h:162
std::vector<int, std::allocator<int> >::~vector stl_vector.h:435
std::_Destroy<std::vector<int, std::allocator<int> > > stl_construct.h:98
std::_Destroy_aux<false>::__destroy<std::vector<int, std::allocator<int> >*> stl_construct.h:108
std::_Destroy<std::vector<int, std::allocator<int> >*> stl_construct.h:137
std::_Destroy<std::vector<int, std::allocator<int> >*, std::vector<int, std::allocator<int> > > stl_construct.h:206
std::vector<std::vector<int, std::allocator<int> >, std::allocator<std::vector<int, std::allocator<int> > > >::_M_erase_at_end stl_vector.h:1518
std::vector<std::vector<int, std::allocator<int> >, std::allocator<std::vector<int, std::allocator<int> > > >::clear stl_vector.h:1248
Teaser::Graph::clear graph.h:189
Teaser::RobustRegistrationSolver::reset registration.h:777

closed 2020-11-29 by cdb0y511