MIT-SPARK/TEASER-plusplus #64: [QUESTION]

LeeCH-736, 2020-11-20:
I read excellent papers “TEASER: Fast and Certifiable Point Cloud Registration”, I have configured your code https://github.com/MIT-SPARK/TEASER-plusplus.
   However, how can I get the matching results (as shown in Fig. 8 and Fig. 18 of the above paper), which with RGB-D images? With the teaser_python_fpfh_icp, I can only get point cloud registration results without RGB information. Can you tell me the matching steps for the RGBD results (Fig. 8 and Fig. 18 in TEASER: Fast and Certifiable Point Cloud Registration) with you code in https://github.com/MIT-SPARK/TEASER-plusplus. 
 
Looking forward to your reply. Thank you very much.

jingnanshi, 2020-11-20:
Hi @LeeCH-736 , thank you for your interests. Although we use RGB-D datasets in the paper, we do not use the RGB color information in our registration algorithm.

LeeCH-736, 2020-11-20:
Thank you for your quickly reply. 
But how to get/show Fig. 8 and Fig. 18 in paper “TEASER: Fast and Certifiable Point Cloud Registration”? That is, are Fig. 8 and Fig. 18 gotten/shown with your code https://github.com/MIT-SPARK/TEASER-plusplus? If this is true,  what are the processing steps with your code to get/show Fig.8 and FIg.18 ? 
however, with the teaser_python_fpfh_icp, only point cloud can be shown, and  Fig. 8 and Fig. 18 cannot be shown with RGB. Thank you very much.

yuecideng, 2020-11-21:
Maybe you can use some 2D features descriptor (SIFT, SUFT, ORB. etc) to find correspoding points from RGBD images, and use TEASER algorithm to solve the matching result @LeeCH-736 .

jingnanshi, 2020-11-21:
@LeeCH-736 Hi, this repo does not contain the experiment / plotting code for Fig. 8 and Fig. 18.

closed 2020-11-21 by jingnanshi