MIT-SPARK/TEASER-plusplus #66: question for Kinect

adrian11655, 2020-11-22:
Hi,
 so from some time I have been trying to run your Teaser++. I put  RGBD dataset obained by Kinect2 into Teaser++, unfortunately with teaser_python_fpfh_icp, i cannot get the registration results as shown in your paper. 
first,  how to use color informaiton, as can be seen. In your paper, the registration results of RGBD dataset is very perfect.  
second,  can you give me the code or tell me the detailed steps to use teaser_python_fpfh_icp to do pefect registration with RGBD dataset. 

I am using:
Ubuntu 18.04
Kinect 2 camera

thank you very much

jingnanshi, 2020-11-22:
Hi @adrian11655 , thank you for your interest. If you are referring to the 3DMatch experiments, we used 3DSmoothNet instead of FPFH descriptors. In addition, although the datasets are RGBD, we don't use color.

adrian11655, 2020-11-23:
Hi, thanks for your reply. 
However, the authors note in your paper, FPFH feature descriptors [1] are firstly used to establish putative correspondences. Therefore, which method (FPFH/3Dmatch) is used in fact? 
It is strongly suggested to give a demo in GitHub for RGBD，which corresponds to the registration results in your paper. 

Thanks

jingnanshi, 2020-11-23:
I think you are talking about the object localization experiment, not the 3DMatch experiment. In that case, FPFH features are used. This repo is for the C++ implementation only. If you are interested in the specifics of the experiments in the paper, please email us, and we can provide you with the experimental code.

jingnanshi, 2020-11-23:
You might also be interested in this: https://github.com/MIT-SPARK/TEASER-plusplus/tree/master/examples/teaser_python_fpfh_icp

adrian11655, 2020-11-23:
> I think you are talking about the object localization experiment, not the 3DMatch experiment. In that case, FPFH features are used. This repo is for the C++ implementation only. If you are [...]

@jingnanshi  Hi，thanks for your reply.  I am interested in the experiments of RGBD (Figure 8). i have sent you an email. please provide me with the experimental code of Figure 8. My email is adrian1122@yeah.net. 
Thanks.

closed 2020-12-09 by jingnanshi