MIT-SPARK/TEASER-plusplus #68: Error teaser_cpp_fpfh

silviadonzelli, 2020-12-05:
**Have you read the documentation?**

- [x ] Yes
- [] No - then this issue will be closed.

Good morning,
 I have the following problem when I try to start the example Teaser_cpp_fpfh example:
CMake Error at CMakeLists.txt:12 (add_executable):
  Target "teaser_cpp_fpfh" links to target "teaserpp::teaser_features" but
  the target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?
so the make command fails and I see nothing. 
What Could I do?
Thank you in advance

jingnanshi, 2020-12-05:
Did you run make install to install TEASER++?

silviadonzelli, 2020-12-05:
Yes!
The teaser_cpp_ply runs correctly but teaser_cpp_fpfh gives me this error and I don’t know Why. It doesn’t find the teaser_features, while it finds the other without problems. I also tried to reinstall it but it gives me always this error. What could I do?

jingnanshi, 2020-12-05:
Can you try changing `teaserpp::teaser_features` to `teaser_features`? There might be a bug in the CMake script.

silviadonzelli, 2020-12-05:
Thank you for your feedback, Now I changed this line in the CMake script, but, while the cmake .. command runs without problem, when I run the make command I receive the folowing error:
/usr/bin/ld: impossibile to found -lteaser_features
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/teaser_cpp_fpfh.dir/build.make:161: teaser_cpp_fpfh] Error1
make[1]: *** [CMakeFiles/Makefile2:76: CMakeFiles/teaser_cpp_fpfh.dir/all] Error 2
make: *** [Makefile:84: all] Error 2

coreos-xx, 2020-12-07:
Exactly the same problem here. 
I get the same error too! 
usr/bin/ld: cannot find -lteaser_features
collect2: error: ld returned 1 exit status

jingnanshi, 2020-12-08:
Try the following things:
1. run `sudo ldconfig`
2. if the above doesn't fix it, run `locate teaserppTargets.cmake` and see whether the file exists.

silviadonzelli, 2020-12-08:
I tried and `sudo ldconfig ` didn’t solve.
‘Locate teaserppTargets.cmake’ returns the following: home/user/TEASER-plusplus/build/teaser/CMakeFiles/Export/lib/cmake/teaserpp/teaserppTargets.cmake
And usr/local/lib/cmake/teaserpp/teaserppTargets.cmake

coreos-xx, 2020-12-08:
Hi Jing and Silvia,
I tried also `sudo ldconfig` but it didn't compile. 
But i found a work around, 
Go to the teaser built folder and rebuild with 
`cmake .. -DBUILD_TEASER_FPFH=ON && make`
and then 
`sudo make install`
Then the "teaser_cpp_fpfh.cc" should compile. If not try `sudo ldconfig` and then compile again.

silviadonzelli, 2020-12-08:
Thank you very much! It works:)

jingnanshi, 2020-12-08:
Good catch! I forgot to check with you guys about the CMake options.

closed 2020-12-08 by jingnanshi