MIT-SPARK/TEASER-plusplus #69: [QUESTION] error: ‘eigen_assert_exception’ is not a member of ‘Eigen’

chuong98, 2020-12-13:
**Have you read the documentation?**

- [X] Yes
- [] No - then this issue will be closed.

Post your theoretical questions / usage questions here.
I follow the install pipeline, 
```
sudo apt install cmake libeigen3-dev libboost-all-dev
cd TEASER-plusplus && mkdir build && cd build
cmake -DTEASERPP_PYTHON_VERSION=3.6 .. && make teaserpp_python
```
but get following Error
```
[ 60%] Built target pmc
[ 66%] Building CXX object teaser/CMakeFiles/teaser_registration.dir/src/certification.cc.o
In file included from /usr/local/include/eigen3/Eigen/Core:88,

In file included from /usr/local/include/eigen3/Eigen/Core:88,
                 from /home/cybercore/Workspaces/chuong/TEASER-plusplus/build/spectra-src/include/Spectra/SymEigsSolver.h:10,
                 from /home/cybercore/Workspaces/chuong/TEASER-plusplus/teaser/src/certification.cc:15:
/usr/local/include/eigen3/Eigen/src/Core/products/Parallelizer.h: In function ‘void Eigen::internal::parallelize_gemm(const Functor&, Index, Index, Index, bool)’:
/usr/local/include/eigen3/Eigen/src/Core/products/Parallelizer.h:162:19: error: ‘eigen_assert_exception’ is not a member of ‘Eigen’
   if (errorCount) EIGEN_THROW_X(Eigen::eigen_assert_exception());
                   ^~~~~~~~~~~~~
make[3]: *** [teaser/CMakeFiles/teaser_registration.dir/build.make:63: teaser/CMakeFiles/teaser_registration.dir/src/certification.cc.o] Error 1
make[2]: *** [CMakeFiles/Makefile2:564: teaser/CMakeFiles/teaser_registration.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:713: python/CMakeFiles/teaserpp_python.dir/rule] Error 2
make: *** [Makefile:344: teaserpp_python] Error 2
```

My System config:
+ Ubuntu18.4
+ Cuda 10.2

jingnanshi, 2020-12-13:
This seems to be a problem in your Eigen installation. See this: https://forums.opensuse.org/showthread.php/546430-openMVS-failed-to-build-possible-bug-in-Eigen3 and this: https://bugs.archlinux.org/task/68118. You might need to include Eigen yourself, or wait for an update in the Ubuntu repo.

chuong98, 2020-12-15:
Hi thank you.
+ You are right, after installing the new release Eigen3, I can build it:
```
git clone https://gitlab.com/libeigen/eigen.git
cd eigen/
mkdir build
cd build/
cmake ..
sudo make install
```
+ After that, running the test script gives me the error `core_dump`. I follow the issue #1 and it works now:
`OMP_NUM_THREADS=12 python teaser_python_ply.py`

closed 2020-12-15 by chuong98