MIT-SPARK/TEASER-plusplus #70: [Algorithm failure]  serious Memory leak on middle size point cloud, need help.

cdb0y511, 2020-12-16:
Technically, It should not consider as a bug,  more like an algorithm failure. 
I compare Teaser++ with ICP kind registration like the T-RO.
ICP kind method can handle 50k points (nearest point correspondence ) with no problem.
However, Teaser++ crushed with a huge memory leak.
I trace it to 

registration.cc

       computeTIMs(const Eigen::Matrix<double, 3, Eigen::Dynamic>& v,
                                              Eigen::Matrix<int, 2, Eigen::Dynamic>* map)

and find

      Eigen::Matrix<double, 3, Eigen::Dynamic> vtilde(3, N * (N - 1) / 2); 

if N = 50k, vtilde will be 500000^2 / 2.
In the following loop, the memory reached its limit.

I know it does not really need 50k to do the registration.
But I want Teaser to compute the max_clique_, I think the PMC process is better than RANSAC, especially with large number of outliers.

To compute    
 
    max_clique_ = clique_solver.findMaxClique(inlier_graph_);

It needs inlier_graph_, and inlier_graph_  needs  src_tims_map_,right?
Any advice will be helpful
Thanks.

cdb0y511, 2020-12-16:
I realize it needs a lot of memory to create the graph,  a lot of memory to store the edge information.

        // Calculate some important indices
        // For each measurement, we compute the TIMs between itself and all the measurements after it.
        // For example:
        // i=0: add N-1 TIMs
        // i=1: add N-2 TIMs
        // etc..
        // i=k: add N-1-k TIMs
        // And by arithmatic series, we can get the starting index of each segment be:
        // k*N - k*(k+1)/2

jingnanshi, 2020-12-16:
Hi @cdb0y511, thank you for your interest and feedback. By design, TEASER++ is unfortunately not very memory efficient. As you noticed, in order to build the graph we check for all possible combinations of the input correspondences.

I will instead suggest significantly reduce the size of the inputs. You should view inputs to TEASER++ as **3D point correspondences**, not raw 3D points. You should use some feature extraction algorithms to first extract features, then use TEASER++ to solve the registration.

In the paper, we demonstrated the possibility of using TEASER++ without any feature matching: essentially we generate all possible correspondences, and input those correspondences to TEASER++ (and we compare it with ICP-based methods). However, due to the large memory requirement, such all-to-all correspondences are only possible for small point clouds (< 200 points on my machine).

Let me know if you have more questions.

closed 2021-01-15 by jingnanshi