MIT-SPARK/TEASER-plusplus #71: [QUESTION] How to suppress the Debug Info on Console

chuong98, 2020-12-24:
**Have you read the documentation?**

- [x] Yes
- [ ] No - then this issue will be closed.

Post your theoretical questions / usage questions here.
Is there any way to turn off the output on console, when calling `solver.solve()`. Here is the typical output:
```
Starting scale solver.
Scale estimation complete.
Max core number: 34
Num vertices: 369
*** [pmc heuristic: thread 3]   current max clique = 16,  time = 2.28882e-05 sec
*** [pmc heuristic: thread 1]   current max clique = 17,  time = 3.88622e-05 sec
*** [pmc heuristic: thread 4]   current max clique = 18,  time = 4.79221e-05 sec
*** [pmc heuristic: thread 3]   current max clique = 20,  time = 6.10352e-05 sec
*** [pmc heuristic: thread 3]   current max clique = 22,  time = 0.000130892 sec
*** [pmc heuristic: thread 1]   current max clique = 24,  time = 0.000287056 sec
[pmc heuristic]  mc = 24
Created adjacency matrix in 4.48227e-05 seconds
[pmc: initial k-core pruning]  before pruning: |V| = 368, |E| = 10452
[pmc: initial k-core pruning]  after pruning:  |V| = 357, |E| = 10149
``` 
Thank you.

jingnanshi, 2020-12-29:
Hi @chuong98 : sorry for the late reply, unfortunately these messages are mostly print by the PMC library. You need to manually remove those print statements. To suppress messages print by TEASER++, pass `ENABLE_DIAGNOSTIC_PRINT=OFF` to CMake.

chuong98, 2020-12-29:
Thabk you.

closed 2020-12-30 by jingnanshi