MIT-SPARK/TEASER-plusplus #72: [BUG] Max clique equal to zero and abort when using descriptors generated by 3DSmoothNet on my own data

IDEA-V, 2021-01-07:
I generate descriptors with 3DSmoothNet on my own data.
The ply contains 271467 points and I randomly choose 5000 keypoints from them
[data.zip](https://github.com/MIT-SPARK/TEASER-plusplus/files/5780614/data.zip)

==================================================
        TEASER++ Python registration example      
==================================================
Reading PLY: [========================================] 100%
Reading PLY: [========================================] 100%
TEASER++ Parameters are: <Params with noise_bound=0.05
cbar2=1
estimate_sccaling=0
rotation_estimation_algorithm=GNC-TLS
inlier_selection_mode=PMC_EXACT
>
Starting scale solver.
Scale estimation complete.
Max core number: 0
Num vertices: 2
[pmc heuristic]	 mc = 0
Max clique lower bound equals to zero. Abort.
Max Clique of scale estimation inliers: 
Clique size too small. Abort.
Max clique size: 0
==================================================
Number of points:  5000
Traceback (most recent call last):
  File "aa.py", line 364, in <module>
    name, scene_path, desc_path, pair[0], pair[1]
  File "aa.py", line 345, in pair_eval_helper
    print("Number of matched correspondences: ", ref_matched_key.shape[1])
IndexError: tuple index out of range

Do you have any ideas about why this is happenning

jingnanshi, 2021-01-07:
@IDEA-V Thanks for reaching out. After you have the keypoints, did you use nearest-neighbor matching to select out matching correspondences based on 3DSmoothNet features? TEASER++ won't do anything if you just straightly feed 5000 keypoints into it.

IDEA-V, 2021-01-08:
> @IDEA-V Thanks for reaching out. After you have the keypoints, did you use nearest-neighbor matching to select out matching correspondences based on 3DSmoothNet features? TEASER++ won't do anything if you just straightly feed 5000 keypoints into it.

I use the script provided by you 
[teaser_python_3dsmooth.py](https://github.com/MIT-SPARK/TEASER-plusplus/blob/master/examples/teaser_python_3dsmooth/teaser_python_3dsmooth.py)

It uses the .npz file generated by 3DSmoothNet after the inference step.

jingnanshi, 2021-01-11:
Check what are the returned values on this line: https://github.com/MIT-SPARK/TEASER-plusplus/blob/4a14392e1315551fd507ec1ecc07743709636099/examples/teaser_python_3dsmooth/teaser_python_3dsmooth.py#L299

They should be containing points fewer than 5000.

IDEA-V, 2021-01-14:
I have found that this is a 3DSmoothNet problem. I have solved it, https://github.com/zgojcic/3DSmoothNet/issues/38#issuecomment-760021491

So I am closing up this issue.

closed 2021-01-14 by IDEA-V

rttariverdi67, 2022-01-26:
dear @IDEA-V 
could you please explain how you encoded .ply file in ASCII instead of binary and removed the color information?

I do as follows but while reading .py file (encoded), I got the error:
```[Open3D WARNING] Read PLY failed: number of vertex <= 0.```

```from plyfile import PlyData, PlyElement```
```name = 'cloud.ply'
pcd = o3d.io.read_point_cloud(binary_ply_dir)
data = np.asarray(pcd.points)
size = data.shape[0]

_array = np.empty(size, dtype=[('vertex_indices', 'f4', (3,))]) #removing the color information
_array['vertex_indices'] = data[:size]
PlyE = PlyElement.describe(_array, name)

PlyData([PlyE], text=True).write('ascii_cloud.ply')# encode .ply file in ascii instead of binary```

ahmadnezam, 2022-12-19:
what is the relation between Max clique and teaser++