MIT-SPARK/TEASER-plusplus #73: Hi, I have intalled the TEASER in the Ubuntu16.04 and there are some problem with python when I installed

AdamPengG, 2021-01-15:
Hi,
I installed the TEASER in the Ubuntu 16.04 and Installed the python3.6.4 and conda2020.
The C++ part is ok.  However, when I installed the python part.
`cmake -DTEASERPP_PYTHON_VERSION=3.6 .. && make teaserpp_python`
Then the terminal shows that :
```
CMake Error at /usr/local/share/cmake-3.19/Modules/FindPackageHandleStandardArgs.cmake:218 (message):
  Could NOT find PythonInterp: Found unsuitable version "2.7.12", but
  required is at least "3.6" (found /usr/bin/python)
```
Which part is wrong?

AdamPengG, 2021-01-15:
I have solved the problem, thank you sir

closed 2021-01-15 by jingnanshi

manhha1402, 2021-02-22:
Hi @AdamPengG,
I have the same issue. How could you solve it? 
Thank you

jingnanshi, 2021-02-22:
@manhha1402 Make sure to enable a conda environment (or any virtual environment) that has the appropriate python interpreter in the PATH variable.