MIT-SPARK/TEASER-plusplus #75: [BUG]

hamid54, 2021-01-26:
Hi,
teaser_python_3dsmooth example does not work. I have a ubuntu 20.04. I followed the given instruction but when I ran teaser_python_3dsmooth, first it asked for pandas and flyfile packages, and then it gave me this error: 
`Traceback (most recent call last):
  File "teaser_python_3dsmooth.py", line 345, in <module>
    scene_path, desc_path, gt_mat, pair[0], pair[1]
  File "teaser_python_3dsmooth.py", line 267, in pair_eval_helper
    frag1 = o3d.registration.Feature()
AttributeError: module 'open3d' has no attribute 'registration'
`

It seems like there is a version conflict there.  By the way, teaser_cpp_ply and teaser_python_ply are running without any problem.  Thank you!

Hamid

**The whole script:**
```
$ conda create -n teaser python=3.6 numpy
Collecting package metadata (current_repodata.json): done
Solving environment: done

** Package Plan **

  environment location: ~/anaconda3/envs/teaser

  added / updated specs:
    - numpy
    - python=3.6

The following NEW packages will be INSTALLED:

  _libgcc_mutex      pkgs/main/linux-64::_libgcc_mutex-0.1-main
  blas               pkgs/main/linux-64::blas-1.0-mkl
  ca-certificates    pkgs/main/linux-64::ca-certificates-2021.1.19-h06a4308_0
  certifi            pkgs/main/linux-64::certifi-2020.12.5-py36h06a4308_0
  intel-openmp       pkgs/main/linux-64::intel-openmp-2020.2-254
  ld_impl_linux-64   pkgs/main/linux-64::ld_impl_linux-64-2.33.1-h53a641e_7
  libedit            pkgs/main/linux-64::libedit-3.1.20191231-h14c3975_1
  libffi             pkgs/main/linux-64::libffi-3.3-he6710b0_2
  libgcc-ng          pkgs/main/linux-64::libgcc-ng-9.1.0-hdf63c60_0
  libstdcxx-ng       pkgs/main/linux-64::libstdcxx-ng-9.1.0-hdf63c60_0
  mkl                pkgs/main/linux-64::mkl-2020.2-256
  mkl-service        pkgs/main/linux-64::mkl-service-2.3.0-py36he8ac12f_0
  mkl_fft            pkgs/main/linux-64::mkl_fft-1.2.0-py36h23d657b_0
  mkl_random         pkgs/main/linux-64::mkl_random-1.1.1-py36h0573a6f_0
  ncurses            pkgs/main/linux-64::ncurses-6.2-he6710b0_1
  numpy              pkgs/main/linux-64::numpy-1.19.2-py36h54aff64_0
  numpy-base         pkgs/main/linux-64::numpy-base-1.19.2-py36hfa32c7d_0
  openssl            pkgs/main/linux-64::openssl-1.1.1i-h27cfd23_0
  pip                pkgs/main/linux-64::pip-20.3.3-py36h06a4308_0
  python             pkgs/main/linux-64::python-3.6.12-hcff3b4d_2
  readline           pkgs/main/linux-64::readline-8.0-h7b6447c_0
  setuptools         pkgs/main/linux-64::setuptools-52.0.0-py36h06a4308_0
  six                pkgs/main/linux-64::six-1.15.0-py36h06a4308_0
  sqlite             pkgs/main/linux-64::sqlite-3.33.0-h62c20be_0
  tk                 pkgs/main/linux-64::tk-8.6.10-hbc83047_0
  wheel              pkgs/main/noarch::wheel-0.36.2-pyhd3eb1b0_0
  xz                 pkgs/main/linux-64::xz-5.2.5-h7b6447c_0
  zlib               pkgs/main/linux-64::zlib-1.2.11-h7b6447c_3

Proceed ([y]/n)? y

Preparing transaction: done
Verifying transaction: done
Executing transaction: done
*
* To activate this environment, use
*
*     $ conda activate teaser
*
* To deactivate an active environment, use
[... 470 lines cut ...]
The following packages will be UPDATED:

  certifi            pkgs/main::certifi-2020.12.5-py36h06a~ --> conda-forge::certifi-2020.12.5-py36h5fab9bb_1

The following packages will be SUPERSEDED by a higher-priority channel:

  ca-certificates    pkgs/main::ca-certificates-2021.1.19-~ --> conda-forge::ca-certificates-2020.12.5-ha878542_0

Proceed ([y]/n)? y

Preparing transaction: done
Verifying transaction: done
Executing transaction: done
(teaser) ~/Projects/Teaser-py/TEASER-plusplus/examples/teaser_python_3dsmooth$ python teaser_python_3dsmooth.py
==================================================
        TEASER++ Python registration example      
==================================================
Traceback (most recent call last):
  File "teaser_python_3dsmooth.py", line 345, in <module>
    scene_path, desc_path, gt_mat, pair[0], pair[1]
  File "teaser_python_3dsmooth.py", line 267, in pair_eval_helper
    frag1 = o3d.registration.Feature()
AttributeError: module 'open3d' has no attribute 'registration'

````

jingnanshi, 2021-01-26:
Hi @hamid54 , thank you for your interest. Can you try with open3d version = 0.10.0?

hamid54, 2021-01-27:
Hi, @jingnanshi thank you for the response! 
I tried with open3d 0.10.0, but nothing changed.

jingnanshi, 2021-01-27:
@hamid54 try 0.9.0.0. This should work.

hamid54, 2021-01-27:
@jingnanshi the first try was 0.9.0.0. Then, after your previous comment, I tried 0.0.10. The result did not change and the error message was the same. Thank you!

jingnanshi, 2021-01-28:
@hamid54 please check again whether you have actually installed open3d 0.9 correctly. On my machine both 0.9 and 0.10 has module registration

Dostoyewski, 2021-03-16:
It will work with the latest version of Open3D.

closed 2021-07-17 by jingnanshi