MIT-SPARK/TEASER-plusplus #76: find_package()

narutojxl, 2021-01-27:
Hi author,
After make installed teaser in `/usr/local/`,  when using in other project CMakeLists.txt with `find_package(teaserpp REQUIRED)`, 
it comes out:
>
	CMake Error at hdl_global_localization_noted/CMakeLists.txt:45 (find_package):
	  By not providing "Findteaserpp.cmake" in CMAKE_MODULE_PATH this project has
	  asked CMake to find a package configuration file provided by "teaserpp",
	  but CMake did not find one.

	  Could not find a package configuration file provided by "teaserpp" with any
	  of the following names:

		teaserppConfig.cmake
		teaserpp-config.cmake

	  Add the installation prefix of "teaserpp" to CMAKE_PREFIX_PATH or set
	  "teaserpp_DIR" to a directory containing one of the above files.  If
	  "teaserpp" provides a separate development package or SDK, be sure it has
	  been installed.
In the `/usr/local/lib/cmake/teaserpp/`, there is no `teaserppConfig.cmake` or `teaserpp-config.cmake`, but `teaserppTargets.cmake` and `teaserppTargets-release.cmake`. How to fix this problem, thanks for your help and time!
>
	/usr/local/lib/cmake/teaserpp/teaserppTargets.cmake
	/usr/local/lib/cmake/teaserpp/teaserppTargets-release.cmake

jingnanshi, 2021-02-02:
Hi @narutojxl , thanks for reporting this issue. Will take a look soon.

narutojxl, 2021-02-03:
Hi @jingnanshi, thanks author :)

StrikerCC, 2021-08-09:
get the same issue, I set teaserpp_DIR to build directory, but there is no teaserppConfig.cmake or  teaserpp-config.cmake also. I'm wondering how your guys run the Minimal C++ example

Aklice-new, 2023-05-29:
你好，请问解决这个问题了吗

LimHyungTae, 2024-04-03:
Please run `sudo ldconfig` after the `sudo make install`!

ivrolan, 2025-05-06:
changing to develop branch fixed it for me. I got the error even running `sudo ldconfig` as suggested in the instructions.