MIT-SPARK/TEASER-plusplus #77: Trying Teaser++ with two .ply files: How can I do?

silviadonzelli, 2021-02-21:
**Have you read the documentation?**

- [x ] Yes
- [] No - then this issue will be closed.

Good Morning,
I'm trying to use Teaser++. I would like to use it with my .ply files.
I opened one file, I removed the Nan values and when I tried it with the T transformation, it worked perfectly.
Now, I want to see the differences between the first and the second point cloud. I removed the Nan values in the second .ply file but when I run the code, it aborts during the method solve.
What could cause this problem?
Thank you in advance
[image]

jingnanshi, 2021-02-22:
So for any ply files, you need to first use some feature matchers to generate correspondences. Then, you pass the correspondences to TEASER++ for solving. Please take a look at the documentation website for examples.

silviadonzelli, 2021-02-22:
Thank you, but I have a question:
Do we generate correspondences in this line, right? With fpfh

> 
  teaser::Matcher matcher;
  auto correspondences = matcher.calculateCorrespondences(
      src_cloud, dst_cloud, *obj_descriptors, *scene_descriptors, false, true, false, 0.95);

jingnanshi, 2021-02-22:
Yes, that looks about right.

silviadonzelli, 2021-02-26:
I found that my problem is in this line of registration.cc: 

>   translation_inliers_ = utils::findNonzero<bool>(translation_inliers_mask_);

If I comment it, the code works fine, but what could be the problem?
Thank you in advance

jingnanshi, 2021-03-03:
Are you feeding the same number of points in source and target point clouds? You should compile it in debug mode to see more detailed error messages.

oldwalkman, 2021-07-01:
> Are you feeding the same number of points in source and target point clouds? You should compile it in debug mode to see more detailed error messages.

Do the number of points in source and target point clouds must be the same?

jingnanshi, 2021-07-01:
@oldwalkman The inputs to TEASER++ should be correspondences between 2 point clouds. In other words, you can have 2 clouds of different sizes, but you have to extract N correspondences, and feed the N pairs of points (as two point clouds / Eigen matrices) into TEASER++.

oldwalkman, 2021-07-02:
> @oldwalkman The inputs to TEASER++ should be correspondences between 2 point clouds. In other words, you can have 2 clouds of different sizes, but you have to extract N correspondences, and feed the N pairs of points (as two point clouds / Eigen matrices) into TEASER++.

Thanks very much, in the other words as I understand, when feeding raw pointcloud without correspondences or feature extraction, the size or the point number of the two poindcloud must be the same.

jingnanshi, 2021-07-02:
@oldwalkman Yes, and you need to manually generate all-to-all correspondences, otherwise it is not likely going to work. By all-to-all correspondences, I mean if you have N points, you need to generate N*(N-1) pairs of points.

closed 2021-07-17 by jingnanshi