MIT-SPARK/TEASER-plusplus #78: PyPi package available and/or planned?

jackjansen, 2021-02-21:
**Have you read the documentation?**

- [x] Yes - insofar as relevant to the question:-)
- [ ] No - then this issue will be closed.

I have looked for a PyPi package, but it seems that it isn't available. So my first question is: did I not look hard enough?

And if I did look hard enough: is a PyPi package on the radar?

And if it isn't: _if_ it turns out that a PyPi package is a reasonable solution for me (i.e. it is not significantly more work for me to do that, as opposed to teaching my CI/CD about all the build steps manually): would you be likely to accept a pull request?

jingnanshi, 2021-02-22:
Hi @jackjansen , no the package is not on PyPi. I'll certainly welcome any pull request that can make this happen :) 
Currently, you can install the python binding locally after compilation, see the documentation for details.

jackjansen, 2021-02-23:
Yes, that is what I am doing at the moment, and that's fine for manual installation, but for our multiplatform CI/CD it's a hassle. I will have a look, time permitting, and I'll give you a pull request if I can make it work.

cc: @nachoreimat

analog-cbarber, 2021-12-15:
Note that due to the dependency on the pmc library which has a GPL3 license, the resulting pip package will also have to have a GPL3 license...

jackjansen, 2021-12-16:
@nachoreimat do you think you could help with creating a PyPI package?

LimHyungTae, 2024-07-31:
Is there any update, guys? @jackjansen ?

jackjansen, 2024-09-16:
@LimHyungTae no updates from our side. Teaser has sort-of moved off our radar. Sorry...

LimHyungTae, 2024-09-18:
No worries, thanks for letting us know!

LimHyungTae, 2024-11-28:
To support Pypi (as a part of [KISS-Matcher](https://github.com/MIT-SPARK/KISS-Matcher)), I've tried to set the pypi pipeline for easy installation.

## What I did

I set `pyproject.toml` file like this, following [`pyproject.toml` in KISS-ICP](https://github.com/PRBonn/kiss-icp/blob/main/python/pyproject.toml):

```
[build-system]
requires = ["scikit_build_core", "pybind11"]
build-backend = "scikit_build_core.build"
[project]
name = "teaserpp_python"
version = "1.1.0"
description = "Python binding for TEASER++"
readme = "README.md"
authors = [
    { name = "Jingnan Shi", email = "jnshi@mit.edu" },
]
requires-python = ">=3.6"
keywords = [
    "Point cloud",
    "Registration",
    "Non-minimal solver",
    "Solver",
]
classifiers = [
    "Intended Audience :: Developers",
    "Intended Audience :: Education",
    "Intended Audience :: Other Audience",
    "Intended Audience :: Science/Research",
    "License :: OSI Approved :: MIT License",
    "Operating System :: MacOS",
    "Operating System :: Microsoft :: Windows",
    "Operating System :: Unix",
    "Programming Language :: C++",
    "Programming Language :: Python :: 3",
    "Programming Language :: Python :: 3.7",
    "Programming Language :: Python :: 3.8",
    "Programming Language :: Python :: 3.9",
    "Programming Language :: Python :: 3.10",
    "Programming Language :: Python :: 3.11",
]
dependencies = [
    "numpy",
]
[project.urls]
Homepage = "https://github.com/MIT-SPARK/TEASER-plusplus"
[tool.scikit-build]
build-dir = "build/{wheel_tag}"
cmake.verbose = false
cmake.minimum-version = "3.16"
wheel.install-dir = "build/pybind/"
[tool.black]
line-length = 100
[tool.isort]
profile = "black"
[tool.pylint.format]
max-line-length = "100"
[tool.cibuildwheel]
archs = ["auto64"]
skip = ["*-musllinux*",  "pp*", "cp36-*"]
[tool.cibuildwheel.macos]
environment = "MACOSX_DEPLOYMENT_TARGET=10.14"
archs = ["auto64", "arm64"]
```

After then, in the root directory, if I run `python3 -m build `, it succeeded like this!

[image]

[... 341 lines cut ...]

setup(
    name='teaserpp_python',
    version='1.0.0',
    author='Jingnan Shi',
    author_email='jnshi@mit.edu',
    description='Python binding for TEASER++',
    # package_dir={'': '${CMAKE_CURRENT_BINARY_DIR}'},
    package_dir={'teaserpp_python': '${CMAKE_CURRENT_BINARY_DIR}/teaserpp_python'},
    packages=['teaserpp_python'],
    # package_data={'': ['*.so']}
    package_data={
        'teaserpp_python': ['*.so'],  
    },
    include_package_data=True
)
```

but it doesn't work :( But as you can see, `*.so` file has generated: `see [43/43] Linking CXX shared module python/teaserpp_python/teaserpp_python.cpython-38-x86_64-linux-gnu.so` in the above info.

By any chance, could you please give me some advice @nachovizzo?

---

The branch can be found here: https://github.com/LimHyungTae/TEASER-plusplus/tree/feature/pypi

EmDash00, 2024-12-02:
@LimHyungTae I'm needing this for one of my own projects. It looks like the `teaserpp_python.cpython-38-x86_64-linux-gnu.so` isn't being built into the wheel despite being built as a part of the build folder. There are other problems with linking against the dynamic link libraries as well. I'm working on fixes for this currently.

EmDash00, 2024-12-02:
@LimHyungTae I got it to work in Linux; however, it should work just as well in Mac OS. I don't think it will work in Windows due to how `.dll` files are linked differently. I'm not sure how to do that and I don't have a large desire to do so at this time.  

You can find it over at [EmDash00/TEASER-plusplus/#feature/pypi@15f2ea0](https://github.com/EmDash00/TEASER-plusplus/tree/15f2ea042da19cb6327aca7dacbaf23c8f919082). You should just be to install it using `pip install git+https://github.com/EmDash00/TEASER-plusplus.git@15f2ea0`.

I removed the minimum python version as that was causing issues with installing on modern python versions. Trying to install with a minimum version of 3.6 made the max supported NumPy version 1.19.5, which doesn't support Python 3.12 or 3.13. My rationale is just that this library doesn't really depend on any specific version of NumPy. Pybind should take care of any intricacies between different versions.

Not sure which specific version you need, but I've tested it from Python 3.7 to Python 3.13. 

@jingnanshi I think this is probably ready for a PR save for the fact it probably doesn't work on Windows. I can probably put a version restriction and we can do it at another time. I'm not sure if it works on 3.6 as the `uv` python build tool and environment manager doesn't work on Python 3.6. I'm not sure how much work we should put into support such an old version. We can probably build a variety of wheels to make the installation easier, but even it should be installable as is now albeit with a long build period.

xiao10ma, 2025-04-13:
So, currently, is it available for downloading with `pip install`?

LimHyungTae, 2025-04-15:
Not yet, but `pip install -e .` in @EmDash00 's repo works.

EmDash00, 2025-04-19:
> So, currently, is it available for downloading with `pip install`?

Not sure when this is going to get merged since it's been a while. @xiao10ma I recommend using the pip install git+https://github.com/EmDash00/TEASER-plusplus/tree/feature/pypi or even the hash if you need something stable for a project. Let me know if there are bugs. 

Any reason for the holdup on the PR @LimHyungTae?